klaasnicolaas/home-assistant-glow #3: Total count is very high

stevetz, 2021-08-05:
Great project Klaas, thanks!

I have a LDR based sensor I made a while ago, it worked great - accurate to my energy supplier's readings - but i wanted to update to use with the new energy integration, so used yours. Now the hourly totals is very high - not reflecting my real use.

I used to use this:
```
sensor:
  - platform: pulse_meter
    id: electricity
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    unit_of_measurement: 'kW'
    name: '${friendly_name} - Electricity Meter'
    internal_filter: 75ms
    filters:
      - multiply: 0.01875

# Daily Count
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Total'
    power_id: electricity

# Enable time component to reset energy at midnight
time:
  - platform: homeassistant
    id: hasstime
```

and now I've got this:
```
sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Electricity Meter'
    icon: mdi:flash-outline
    unit_of_measurement: 'Watts'
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    internal_filter: 75ms
    accuracy_decimals: 2
    filters:
      - multiply: 18.75
    total:
      name: '${friendly_name} - Total Electricity'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        - multiply: 0.001
```

There isn't much documentation about the `total` variable, so I'm not sure if I have it right - but on the face of it, it appears fine ...  clearly something's going wrong! I'll add, in the new configuration the instantaneous `Electricity Meter` still appears correct with what is using power in the house.

klaasnicolaas, 2021-08-05:
The power meter works fine? LDR is known to be less accurate, so I don't recommend using it.

stevetz, 2021-08-05:
Hi Klaas, yes the meter has been fine, it usally comes out within 1/100th kWh of the utilities daily readings, so I'm happy with that!

I think I've solved the problem(!), but will need to test a while. If it works, it might be something to update in your yaml notes to help future users? and I will mention to esphome to clarify in their docs too.

On the [esphome docs](https://esphome.io/components/sensor/pulse_meter.html#re-setting-the-total-pulse-count), it has this note at the bottom:
```
Note
This value is the raw count of pulses, and not the value you see after the filters are applied.
```

So the filter in `total` should in fact be:
```
sensor:
  - platform: pulse_meter
 ...
    total:
    ...
      filters:
# multiply value = (1 / imp value)
        - multiply: 0.0003125 # My meter is 3200 imp/kWh
```

Does that make sense?

I'll report back tomorrow once its run for the last 12 hours of the day and my energy provider has updated my daily readings.

stevetz, 2021-08-05:
OK this definitely fixed my issue. it was the multiply amount. I've made a pull request to add a note before that line with the calculation. Also another minor PR for something else.

Great work Klaas!

closed 2021-08-05 by stevetz

klaasnicolaas, 2021-08-05:
Thanks for looking into it further! I'll test and merge your PR later. We just had a power outage here so I have something else on my mind now 😅