klaasnicolaas/home-assistant-glow #4: Cant see any entity or values

Snobs76, 2021-08-05:
I`ve got  a working pulse counter and in the past I send the values via mqtt.
Now I copyed your yaml file and do the changes for my secrets etc. At the webinterface i can see all the values and their updates. But I`ve got no entity at ha nothing at developer console, What I`m doing wrong i want to use the new energy function.

Please help.
Best regards
Sascha

`---
#
substitutions:
  device_name: pulse_counter
  friendly_name: House
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO0
  status_led: GPIO2

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home_assistant_glow"
    version: "1.1.1"
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${device_name}'
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  #password: !secret esphome_api_password

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

web_server:
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO5
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} - ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} - IP Address"
      icon: mdi:wifi
    ssid:
      name: "${friendly_name} - Connected SSID"
      icon: mdi:wifi-strength-2

sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'Watt'
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    on_value:
      then:
        - light.turn_on:
            id: led_red
            flash_length: 400ms
    filters:
      # multiply value = (60 / imp value) * 1000
      - multiply: 12
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        - multiply: 0.001`

[image]

klaasnicolaas, 2021-08-05:
If all is well, Home Assistant will automatically recognize a new device and ask if you want to add it.

Snobs76, 2021-08-05:
Thank you for your replay. Thats the problem, HA did not recognize a new device, I delete the old ESP Device and restartet my HA Host not only software. No new device. Any ideas ?

Snobs76, 2021-08-05:
I figured out, he won't find the devices automatically for some reason. I have now set it up manually and watch out, everything I need is there. Thanks for the yaml and your support.

closed 2021-08-05 by Snobs76