klaasnicolaas/home-assistant-glow #5: German Meter WS74 (WS7412.1) works with Infrared LED - Will it also work?

SmartLiving-Rocks, 2021-08-05:
Fantastic job! That is really great, what you have created!

Now I also need to monitor my energy usage. So I have checked my meter for an LED, but it was not visible (at least for the human eye). So I've had a look at the manual and it says it is an infrared LED (5000 Imp/kWh). I am wondering if the photodiode will also work, or do I need a different one?

I have attached a photo of my meter as well as a screenshot of the manual. [Here you can find the full manual in German](https://www.dzg.de/fileadmin/dzg/content/downloads/produkte-zaehler/ws74/dzg_ws74_Bedienungsanleitung_01.pdf)

Thanks in advance and have a nice day! 

[image]

[image]

[Manual in German](https://www.dzg.de/fileadmin/dzg/content/downloads/produkte-zaehler/ws74/dzg_ws74_Bedienungsanleitung_01.pdf)

OttoWinter, 2021-08-05:
Hi, I don't _think_ the photodiode module listed in the README would work here (but I also didn't find any datasheet for it)

Generally the LED will probably be emitting in the range of 800-1000 nm (commonly used in TV remotes and such)

But you can instead look for phototransistors like this one: https://m.banggood.com/PT908-7C-Side-phototransistor-1_5mm-Transparent-860nm-400-1100nm-Photosensitive-Receiver-p-1752121.html?rmmds=search&act_poa=SKUG13853&cur_warehouse=CN - the important bit is that the datasheet wavelength contains the LED wavelength.

You can validate it's likely in the 800-100 nm range by pointing your smartphone camera at the LED - if you see it blink then it's good.

klaasnicolaas, 2021-08-05:
I think the Glow works fine on this meter, however you have a different impulse rate so the multiply for `power` in filter will be different. In your case: `12`.

SmartLiving-Rocks, 2021-08-08:
Thanks for your answer. Regarding my own research, I think my meter has an extended data interface - [eHZ: elektronische Haushaltszähler ](https://de.wikipedia.org/wiki/Smart_Message_Language)

[image]

You can read it out with a [IR-read/write head](https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf). It has two main optical components:
1. [sfh487-2 ](https://www.mouser.de/ProductDetail/OSRAM-Opto-Semiconductors/SFH-487-2?qs=nTDll3UaDK6ZS6zeY7T0tw%3D%3D)
2. [sfh309fa-4](https://www.mouser.de/ProductDetail/OSRAM-Opto-Semiconductors/SFH-309-FA-4?qs=nTDll3UaDK5PTBKWZyYpiQ%3D%3D)

_[Please see here a detailed technical description of the reader](https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-ttl-ausgang) as well as a [How-to Video](https://www.youtube.com/watch?v=5rWxMc3Q3wU&t=0s)_

There is also a version for [ESPeasy](https://wiki.volkszaehler.org/hardware/controllers/espeasy) but I definitely prefer ESPhome! Is that possible and how much work is it @OttoWinter ?

But to keep it simple in the first step, I would like to try out (and support your work) your solution @klaasnicolaas ? I think Paulus mentioned that there is a shop, and we can buy it for around 20$. Where do I find it?

Thanks in advance you two and have a nice Sunday!

klaasnicolaas, 2021-08-08:
Number 5 is indeed a separate interface that is also used by people from your energy company to configure your meter etc. But the Glow will work fine on number 8 just adjust the `multiply` values for your 5000 imp/kWh rate.

I don't sell the Glow in a webshop. The one that Paulus has often mentioned is Marcel Zuidwijk's [P1 smart reader](https://github.com/zuidwijk/dsmr), which you plug into the serial port of your meter. But for those who really want to support/appreciate me in a financial form, it has recently become possible to sponsor me via Github Sponsors.

fourtrax01, 2021-08-13:
I can confirm that it does work with the ir led on my sensus meter after adjusting the potentiometer on the photodiode.     I don't know my pulse rate on my meter yet so still have to figure that out.

klaasnicolaas, 2021-08-13:
@fourtrax01 IR LED? I think you mean just a normal LED. The rate should be listed somewhere, if you need help open a separate issue with information / photo of your meter and we will look for it.

mcguenther, 2021-08-13:
I can also confirm that a regular photodiode works with the WS74. I bought a [board with the LM393-N on ebay ](https://www.ebay.de/itm/284109007684) which specifies the range of the photodiode as 400 to 640 nm. Although I cannot see the pulse with my bare eye, my set-up has worked ok for a day now.

yan14, 2021-08-17:
> Number 5 is indeed a separate interface that is also used by people from your energy company to configure your meter etc. But the Glow will work fine on number 8 just adjust the `multiply` values [...]

if i get the math right the multiply must be changed from 0.06 to 0.12 and for the total from 0.001 to 0.002, right?

yan14, 2021-08-17:
> Thanks for your answer. Regarding my own research, I think my meter has an extended data interface - [eHZ: elektronische Haushaltszähler ](https://de.wikipedia.org/wiki/Smart_Message_Language) [...]

The interface 5 uses SML. ESPHome has no libary for this. See this feature request:
https://github.com/esphome/feature-requests/issues/1041

klaasnicolaas, 2021-08-17:
> if i get the math right the multiply must be changed from 0.06 to 0.12 and for the total from 0.001 to 0.002, right?

@yan14 If you use the latest version you don't even have to calculate anymore, just enter the imp/kWh rate value at the top 😉

closed 2022-01-27 by github-actions[bot]