klaasnicolaas/home-assistant-glow #7: How to use this and use as little power as possible? Deep sleep? Disconnect wifi?

hairychris, 2021-08-05:
I'm trying to use a large battery to get some readings without having to touch the mains wiring.

Is it possible to use any sleep modes to preserve power? Could we wake when the impulse comes in and increase the counter? If not can we disconnect the wifi and only sync once per minute for example?

Any other suggestions more than welcome.

klaasnicolaas, 2021-08-05:
I don't think this is the right place to ask this question, what you want is far from what Glow is made for. Or certain features that are more suitable for the [ESPHome](https://github.com/esphome/feature-requests) repository.

hairychris, 2021-08-05:
I guess it depends on how you look at what glow is.

If it is a one stop solution for people using home assistant to monitor their non P1 meter then I disagree. There will be many others attempting to use this with battery power as they lack the knowledge or experience to wire a plug in their meter cupboard. This issue might require an upstream issue to be opened with ESPHome but it is still relevant.

If glow is just an example of what can be done on ESPHome then it is different.

I'd hoped the glow project could become a meeting place for the many people using similar solutions so we can work on a better solution together. To me a power meter that uses less power is a good thing.

hairychris, 2021-08-05:
This is already implemented in ESPHome if I'm not mistaken:

https://esphome.io/components/deep_sleep.html

klaasnicolaas, 2021-08-05:
You do have a good point there, I'm going to try to get something working with deep sleep. For those who do not have a socket and do not have a P1 port, this can be a solution. If you already have something working in the meantime, please open a PR 😉

hairychris, 2021-08-05:
Thanks. Unfortunately I do not have something working. Deep sleep wasn't implemented in ESPHome last time I tried.

I've got an ESP32 on order (I went for Firebeetle as they seem to be decent power usage wide) so I'm happy to help where I can when that arrives. I don't understand if the pulse counter will increment if its using the same pin as the deep sleep wake up pin and that pin goes high. Perhaps it will depend on how long the pin is high for?

https://diyi0t.com/reduce-the-esp32-power-consumption/

hairychris, 2021-08-05:
If we weren't using ESPHome this is how we would do it:

https://github.com/espressif/esp-idf/tree/master/examples/system/ulp_fsm/ulp

klaasnicolaas, 2021-08-06:
We talked about it internally and we actually still have our doubts whether this will really work. During sleep nothing will be counted and continuously coming out of sleep also costs power.

Ideally you would like it to keep counting, but send an update once every X time. That's not something I can realize and is more with ESPHome than in this repository.

But if you have the stuff, go ahead and test it. For now I'm closing the issue.

closed 2021-08-06 by klaasnicolaas

hairychris, 2021-08-06:
I understand but just as a point of note if you look at this sample code you will see it is quite clearly possible to count pulses whilst in ULP. You are right that this would likely need to be done at the ESPHome level. I'm going to try and raise this over there and then see if we can revisit this. 

https://github.com/espressif/esp-idf/tree/master/examples/system/ulp_fsm/ulp