klaasnicolaas/home-assistant-glow #10: Any chance it will work with Flame sensor

jacauc, 2021-08-06:
I have a HW484 flame sensor, I'd like to use if possible. Don't have a photodiode handy. 
Any chance it would work with this sensor?

klaasnicolaas, 2021-08-06:
How does this sensor detect a flame? Is that based on light?

jacauc, 2021-08-06:
I would imagine so...never used it before

https://m.banggood.com/20pcs-KY-026-Flame-Sensor-Module-IR-Sensor-Detector-Temperature-Detecting-p-1405151.html?utm_source=googleshopping&utm_source=googleshopping&utm_medium=cpc_organic&utm_medium=cpc_bgs&gmcCountry=ZA&utm_content=minha&utm_content=sandra&utm_campaign=minha-za-en-mb&utm_campaign=sandra-ssc-za-all-0628-20LP&currency=ZAR&cur_warehouse=CN&createTmp=1&ad_id=445211076419&gclid=Cj0KCQjwu7OIBhCsARIsALxCUaNGYsYrl4CLnHplaLRaf9Vt5khMB2VQx3UFr1y0Zi-t7UaUGJF9NRAaAp7HEALw_wcB

jacauc, 2021-08-06:
Come to think of it, it's likely sensing IR (heat) and not light

klaasnicolaas, 2021-08-06:
Then I don't think this sensor will work

jacauc, 2021-08-06:
Do you know of any other typical arduino sensors that will cut it? I have a bundle of sensors, one of those starter kits. Don't have the one in the spec but looking for something that'll work since I don't have easy access to parts or shipping.

klaasnicolaas, 2021-08-06:
Anything involving a photodiode or phototransistor will work. Don't currently have any other specific types except the one used in the current version of Glow :(

jacauc, 2021-08-06:
Thanks! Will experiment!

klaasnicolaas, 2021-08-06:
That also works, but an LDR is known for being inaccurate (increased chance of false readings). So as mentioned in the readme, I don't recommend this one, but you can try it.

jacauc, 2021-08-06:
Will have to try the ldr. All I have at the moment. Appreciate it

stevetz, 2021-08-06:
> Will have to try the ldr. All I have at the moment. Appreciate it

I have this working with an LDR. I have the option for an internal filter set to 75ms - took a bit of trial and error between 30-100ms to make sure it only counts each pulse once. Now it is within 1/100th kWh for a day to my energy company’s reading- a rounding error, i’m more than happy to be a few Wh out!
```
sensor:
  …
  internal_filter: 75ms
```

jacauc, 2021-08-07:
> I have this working with an LDR. I have the option for an internal filter set to 75ms - took a bit of trial and error between 30-100ms to make sure it only counts each pulse once. Now it is within [...]

Is that using an Analog sensor? My LDR has only 3 pins

stevetz, 2021-08-07:
I'm not sure what kind of LDR you have @jacauc - I've only ever seen them with 2 pins, ie as a simple resistor.

I found it works well with just the internal pull-up rather than another resistor to ground.

Mine is something like this from Amazon: https://www.amazon.co.uk/dp/B00NXW9WZ6/ref=cm_sw_r_cp_api_glt_fabc_F44MT88Q25V56NXA9VJR

jacauc, 2021-08-07:
Sorry, should have clarified.

I have this type:
https://www.aliexpress.com/i/32900363698.html

henkburnett, 2021-08-09:
I am using one of the LDR modules and I am finding that the LDR seems to consistently detect the LED flashes from my utility meter, but that my attached LED often misses flashes. Would this unit be stable enough?
https://www.botshop.co.za/product/ldr-on-pcb-light-sensor/