klaasnicolaas/home-assistant-glow #11: Battery Powered Glow

QGgWW2WhH, 2021-08-06:
Hello, 

I have no option of plugging in a USB charger next to meter (shared hallway of a multi-tenant building). 

I was thinking a energy efficient Lorawan-Solution which is battery-based might be interesting? 

Any thoughts on this?

klaasnicolaas, 2021-08-06:
I read that some also use a power bank for power supply, maybe that is also within your options.

JonasPed, 2021-08-06:
I am in the same situation where I do not have any plug next to the meter. Have you tried to measure how much power it draws? I am thinking of how realistic it is to use an power bank for instance.

klaasnicolaas, 2021-08-06:
Noy yet, but I've the stuff to try this out 😉

JonasPed, 2021-08-06:
That would be awesome. Perhaps over one hour or something like that. That way we can get an realistic idea of the consumption.

Ulrar, 2021-08-07:
I've got mine running on a power bank and it lasted ~12 hours. I'll try it again today but I'd be surprised if I got much more than that.
Maybe if you can find one without a bunch of LEDs it'll last a bit longer but I don't think that's going to be a great solution sadly.

bencorrado, 2021-08-07:
This is going to be hard to accomplish on an ESP based part, they are just not designed for battery opereated solutions over a long period of time.  

Check out mysensors and the pulse power project.  There are some notes on how a guy got 12 months on two AA batteries with it.   [https://www.mysensors.org/build/pulse_power](https://www.mysensors.org/build/pulse_power)

webworxshop, 2021-08-07:
@bencorrado Whilst it's true that the ESP32 does use much more power than the ATMega processor used in the mysensors build, it's still possible to achieve battery powered operation, through the use of sleep modes (which is actually what the mysensors code you linked is also doing).

There are some really good tips for reducing the power use available here: https://savjee.be/2019/12/esp32-tips-to-increase-battery-life/

For solar power usage the following video is my go to resource: https://www.youtube.com/watch?v=37kGva3NW8w

The following calculator is also useful for calculating the battery life of IoT devices: https://www.of-things.de/battery-life-calculator.php, as long as you have a multimeter for measuring your current usage.

Unfortunately, it's not possible to run the esphome pulse counter and use the ESP32's deep sleep mode, since the processor needs to be awake to observe each pulse. It should be possible to use a different sleep mode (using more current than deep sleep), which would allow the ESP32 to react to an external interrupt. However, I'm not sure how good the support for these modes is in esphome. There is a good overview of the ESP32 sleep modes here: https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/

In theory you could use the ULP co-processor (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/ulp.html) to count the pulses while the main ESP32 cores are asleep. Again, this isn't currently supported in esphome.

I would be great to improve the low power support in esphome. However, low power operation is generally regarded as one of the harder problems in embedded systems, so it's not surprising that we have a way to go.

Ulrar, 2021-12-08:
I wonder if ESP Now would be an option for this ?
Supposedly it uses about one fifth of the power used for regular wpa2 connections, maybe it'd make the battery last a few days instead of hours ?
I say this having never used it, just a thought.

btwebz, 2022-01-07:
I get on average 3 to 3.5 days of operation using 10000 mAh powerbanks.

closed 2022-02-14 by github-actions[bot]