klaasnicolaas/home-assistant-glow #12: Question on units

Ulrar, 2021-08-06:
Hi,

Unrelated to my other issue about powering it (found out that there's a way to force on my battery), I'm not sure the values it's reading are correct.

What's the unit for the power consumption, it says "Watt" but is that per hour ?
I'm getting values in the 3000 watts, and I can't think of what would be drawing that much right now.

Is the default config in the repo good for 1000 imp / kwh ?

Thanks

klaasnicolaas, 2021-08-06:
`Watt` is now `W`, it stands for the current power consumption.
And yes the default config is based on 1000 imp/kWh.

Ulrar, 2021-08-06:
So it's the actual measurement at the time, that makes sense. The numbers actually came down since and they now make more or less sense to me. I still don't know if the measurement is accurate but it looks reasonable.
I wonder what could have been drawing 3000 earlier.

I suppose the only way to confirm the readings would be read the meter, wait an hour, read the meter again and see if it matches up ?

KShips, 2021-08-08:
> So it's the actual measurement at the time, that makes sense. The numbers actually came down since and they now make more or less sense to me. I still don't know if the measurement is accurate but [...]

That would work, also our meter has select button, push it until current power usage is displayed. Compare that to what HA is reading.

My big issue now is that the meter pulses for any power flowing through it, having solar means it could be importing or exporting... not sure how I'm going to get around that.

klaasnicolaas, 2021-08-08:
@KShips In combination with solar panels is a challenge, you can put a separate meter on your solar panels and then start calculating. But you also have 2 different intervals.

@Ulrar Have we already fully answered your question? If so, I'll close this issue.

closed 2021-08-10 by klaasnicolaas