klaasnicolaas/home-assistant-glow #8: How to power it

Ulrar, 2021-08-06:
Hi,

I had everything (except the LED) on hand so I gave this a try, but I don't know how to power it.
I tried plugging it in to a usb battery (the kind you'd charge you phone with) but it keeps turning off, I think it's not drawing enough for the battery to stay on. Maybe I need a dumber battery.

What's a good way to power this ? Any way to not have to recharge a battery regularly, if I find one that works ?

As a bonus question, is there any risk of conflict with the guy who comes to read the meter from time to time, could they have an issue with finding this attached to the meter ? The meter LED is just below the screen, it's still readable but it's definitely a bit in the way.

Thanks

klaasnicolaas, 2021-08-06:
Personally I have power in my meter cupboard so I can just use a power adapter with a long USB cable. It seems to me that a power bank should also work fine, the only question is how long can the glow continue to work with one battery charge 🤔 

As for the electricity guy, it kind of depends on what he needs to read. In our country it is remote and in others it is the IR sensor. Can you share which meter you have?

Ulrar, 2021-08-06:
Hi,

I'm in Ireland, the meters are outside here because they have to be able to access them at anytime to read the values. It's still readable, I'm just worried they might consider this tampering if they don't understand what it is, and since it covers a bit of the screen it's hard to miss.

As for the battery I'd hope it could run for a while, but I'm afraid I'll have to buy a dumber battery to find out. I wonder if this could be made autonomous with a small solar panel, if that's even a thing ? I don't know how much ESP32s can use overnight.

klaus1956, 2021-08-06:
Usual Powe Banks switch off if energy is too low - I ordered the harware today and I have a Fritzbos Router close by where I want to use one of its USB-Ports for power.
If it is outside and no power is close by a small solar pannel with a buffer battery might be a solution

klaasnicolaas, 2021-08-06:
I think your electrician isn't too happy about this (because it might get in the way) 😅 , suspect he needs to be at the IR section (top left) or be able to read the meter reading on your display.

Ulrar, 2021-08-06:
Yeah that's what I'm afraid of, that led is in a really awkward place. Maybe I should just go with a CT clamp instead

klaasnicolaas, 2021-08-06:
What you can do is use another phototransistor, without a whole board you can make the piece that you stick on your sensor smaller and then it should not be a problem.

For a V2 of Glow I want to see if we can make something with a phototransistor, so that would be a solution in the future.

Ulrar, 2021-08-06:
Actually I guess I could just take it off the board and solder some wire between it and the board, that should do it.
I'd just need to come up with some half decent way of housing it, maybe just some black tape directly on the led, or try to cut the top bit of your stl although I'm terrible at cad

LarsNorgaard, 2021-08-10:
There's 230V AC to 5V DC power supplies, that you can mount in a din rail. MEANWELL makes a bunch.

klaus1956, 2021-08-10:
I decided to integrate a standard Socket to the electric cabinet (ordered today) so I can install standard USB charger

danmed, 2021-09-27:
> I decided to integrate a standard Socket to the electric cabinet (ordered today) so I can install standard USB charger

I'm in the UK and have the same issue.. how / where are you getting a socket you can integrate in to the cabinet?

LarsNorgaard, 2021-09-27:
> > I decided to integrate a standard Socket to the electric cabinet (ordered today) so I can install standard USB charger
> 
> I'm in the UK and have the same issue.. how / where are you getting a socket you can integrate in to the cabinet?

Google "Din Rail Socket" Should give you some results :)

danmed, 2021-09-27:
> > > I decided to integrate a standard Socket to the electric cabinet (ordered today) so I can install standard USB charger
> > 
> > 
> > I'm in the UK and have the same issue.. how / where are you getting a socket you can integrate in to the cabinet?
> 
> Google "Din Rail Socket" Should give you some results :)

I'm likely missing something.. this is what my cabinet looks like outside.. i'm guessing there's nothing here i can attach a din rail socket to?
[image]

klaus1956, 2021-09-27:
[image]
Filing carbinets seem to be quite different from location to location

klaasnicolaas, 2021-11-18:
Could we close this issue?

Ulrar, 2021-11-18:
I suppose so, guess unless you're lucky enough to have a socket in there, there's no easy answer. Closing

closed 2021-11-18 by Ulrar