klaasnicolaas/home-assistant-glow #9: Question about Energy total

Snobs76, 2021-08-06:
My Sensor run since round about 24h. The Energy usage every second looks like before with my own configuration pulse-counter. But the total energy consumption is now doubled. So at night, I have an energy consumption round about 250W every second tell the sensor  but at HA ist shows it should be 1.5kwh. Are these values correct it feels much to high ?

Do you have any idea ?

Best regards
Sascha

klaasnicolaas, 2021-08-06:
This question is difficult to answer, as in I don't know your situation and therefore cannot say whether the sensor is right/wrong 😅

Snobs76, 2021-08-06:
Some example: 
If i have a power consumption 250W, then i need 4 hours to get 1kwh, right ?
But the energy total sensor calculates 1.5kwh in one hour with this 250W that I use at night.
There is anywhere a calculation error inside the sensor. Actually he said I have used 60kwh at 24h and this is to much, or for today until now 32.6kwh. I´m not the energy saver and i have 12-15kwh at a day (Home Office etc.).

But where is my error, I have 5000 Imp per Kwh, this is is in yaml factor 12 and looks good, but not the energy total.

klaasnicolaas, 2021-08-06:
What if you change the `multiply` at total to: `0,0002`?

Snobs76, 2021-08-06:
`---
#
substitutions:
  device_name: pulse-meter
  friendly_name: House
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO0
  status_led: GPIO2

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home_assistant_glow"
    version: "1.1.1"
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: on

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${device_name}'
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_api_password

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

web_server:
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO5
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} - ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} - IP Address"
      icon: mdi:wifi
    ssid:
      name: "${friendly_name} - Connected SSID"
      icon: mdi:wifi-strength-2

sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'Watt'
    
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    on_value:
      then:
        - light.turn_on:
            id: led_red
            flash_length: 400ms
    filters:
      # multiply value = (60 / imp value) * 1000
      - multiply: 12
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        - multiply: 0.001`
       
This is the actually flashed code

Snobs76, 2021-08-06:
I will try it

Snobs76, 2021-08-06:
Wrong direction now it counts 0.450Kwh in ~5 Minutes at 450W consumption

Snobs76, 2021-08-06:
Damn idea.... do you have 10000 Imp per Kwh ?
If yes then I need instad of 0.0001 0.00005 for my 5000 Imp per Kwh
Is this possible ?

klaasnicolaas, 2021-08-06:
The YAML file is based on 1000 imp/kWh. Based on the note from @stevetz I was thinking the value for you should be `0.0002` 🤔

Snobs76, 2021-08-06:
I will give your number another try :D

klaasnicolaas, 2021-08-06:
We have been (re)calculating again and the values `12` and `0.0002` must be correct. If the values are still not correct, it may be due to the hardware. A sensor that is too sensitive, causing it to count 1 pulse twice or missing a pulse.

Snobs76, 2021-08-06:
The 0.0002 looks good, I´m testing but how do I calculate the 0.0002.

klaasnicolaas, 2021-08-06:
The calculation: https://github.com/klaasnicolaas/home-assistant-glow/blob/master/home_assistant_glow.yaml#L111

Snobs76, 2021-08-06:
Great, many thanks. Now my values are correct.
I have to buy a coffee/beer for man :D

closed 2021-08-06 by Snobs76

miv777, 2021-08-06:
Please, tell me also what the settings will be when the value is 3200 imp/kWh
Thx 😁

Snobs76, 2021-08-06:
Look at post from @klaasnicolaas  there all the math that you need.

klaasnicolaas, 2021-08-06:
@Snobs76 Maybe I should setup the sponsorship option in Github 😅 Beer and coffee is not my taste, that's why I have a **buy me a fanta** button on my [HA config page](https://github.com/klaasnicolaas/Student-homeassistant-config) 🤣 

@miv777 for 3200 imp/kWh it would be: `18.75` and `0.0003125`.

stevetz, 2021-08-06:
Klaas I wonder if there is something further you could add to clarify the sums - I thought the comment i put in the PR would be enough -  but seems a few issues with the multiply figures...

Maybe an addition to the readme, something like:

> You will need to know the impulses per kWh, or imp/kWh of your meter. Most meters flash at 1000 imp/kWh.
> Use this figure in the `multiply` variables for both the main `sensor` and the `total`.

(not sure if the e.g. is necessary?)

> e.g. for a meter with 3200 imp/kWh (for 'SMART' branded meters in the UK)
> the multiply values will be:
> ```
> sensor:
>   - platform: pulse_meter
>     ...
>     filters:
>       - multiply: 18.75 **# = (60 / 3200) * 1000**
>     total:
>       ...
>       filters:
>         - multiply: 0.0003125 **# 1 / 3200
> ```
> 
> For more information see: https://esphome.io/components/sensor/pulse_meter.html#converting-units

To be sure, I don't know whether its worth the additional language - or whether once I've submitted the PR to the esphome doc (plan to do this weekend) that might be enough?

As I'm here - I also wanted to ask if you thought of adding this to the 'cookbook' on esphome?

klaasnicolaas, 2021-08-06:
I think it's wise to put some information in the readme, even if this information will soon also be on the esphome website.
As for the entry in cookbook, I'll come back to that later.