klaasnicolaas/home-assistant-glow #13: consumption data is not saved after reboot

Dilkins85, 2021-08-07:
after reboot, all readings are reset to zero, is this normal? and for some reason, the new integration of energy does not see the readings (. is it so for everyone?

klaasnicolaas, 2021-08-08:
It is normal for the values to reset after rebooting or flashing, if you would like to keep the values, you could use [utility_meter](https://www.home-assistant.io/integrations/utility_meter/). But this behavior is not a problem for the energy dashboard calculatings.

If you don't have any readings in the dashboard, check if the entity is included in your recorder.

jacauc, 2021-08-11:
Very new to esphome, would you mind giving me some more pointers on how to substitute utility meter into the code?

klaasnicolaas, 2021-08-11:
I don't really understand your question, this is not code that you put in esphome but an integration that you use in Home Assistant.

jacauc, 2021-08-12:
Oh! Thank you!

klaasnicolaas, 2021-08-16:
Has your problem already been solved? Otherwise I would like to close this issue.

closed 2021-11-18 by klaasnicolaas