klaasnicolaas/home-assistant-glow #15: Leaps in readings

PhantoMMM, 2021-08-08:
Good afternoon! Sometimes the power consumption sensor transmits very high values (more than 150,000 watts), and then shows the correct data again. This happens several times a minute. It can be fixed?

PhantoMMM, 2021-08-08:
https://postimg.cc/gallery/sjyxt4q

KShips, 2021-08-08:
I had this problem, it was due to too much light from outside getting to the sensor. 2 options are to either block out more of the sensor or turn the trimpot sensitivity down, combination of both and bit of testing seems to have got mine working well.

PhantoMMM, 2021-08-08:
I set the sensitivity to the minimum level. Maybe you can apply some kind of software filter? My electricity meter has 6400 flashes per kilowatt, if that's important. I am using Esp8266 module.

klaasnicolaas, 2021-08-08:
It sounds like the sensor is measuring ghost flashes, which happens when the time between 2 pulses is very short. Could you share how your sensor is mounted on your meter? And have you adjusted your multiply values needed for your 6400 imp/kWh rate?

PhantoMMM, 2021-08-08:
My config looks like this: 
---
#
substitutions:
  device_name: home_assistant_glow
  friendly_name: House
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO12
  status_led: GPIO5

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home_assistant_glow"
    version: "1.1.2"
  platform: esp8266
  board: nodemcu

wifi:
  ssid: Vladofon
  password: VlAdIsLaV091078

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${device_name}'
    password: esphome291

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 091078091078

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

web_server:
  port: 80
  auth:
    username: admin
    password: 091078

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO4
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} - ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} - IP Address"
      icon: mdi:wifi
    ssid:
      name: "${friendly_name} - Connected SSID"
      icon: mdi:wifi-strength-2

sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    on_value:
      then:
        - light.turn_on:
            id: led_red
            flash_length: 400ms
    filters:
      # multiply value = (60 / imp value) * 1000
      - multiply: 9.375
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        - multiply: 0.00015625

The sensor was previously used with Mysensor and showed correct values.

ArkhonTiger, 2021-08-10:
I'm also getting this. Seems to occur with any level of usage, not just high usage. Log shows a single spike of 360000.00000W each time and then drops back to normal on the next read. (and it's always 360000)

`[10:51:00][V][sensor:035]: 'House - Total energy': Received new state 366.000000
[10:51:00][VV][sensor.filter:013]: Filter(0x3ffb9444)::input(366.000000)
[10:51:00][VV][sensor.filter:020]: Filter(0x3ffb9444)::output(0.366000) -> SENSOR
[10:51:00][D][sensor:127]: 'House - Total energy': Sending state 0.36600 kWh with 3 decimals of accuracy
[10:51:00][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 2690257735
  state: 0.366
  missing_state: NO
}

[10:51:00][V][sensor:035]: 'House - Power consumption': Received new state 6000.000000
[10:51:00][VV][sensor.filter:013]: Filter(0x3ffb928c)::input(6000.000000)
[10:51:00][VV][sensor.filter:020]: Filter(0x3ffb928c)::output(360000.000000) -> SENSOR
[10:51:00][D][sensor:127]: 'House - Power consumption': Sending state 360000.00000 W with 0 decimals of accuracy
[10:51:00][D][light:093]: 'Red' Setting:
[10:51:00][D][light:125]:   Flash Length: 0.4s
[10:51:00][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 2274151077
  state: 360000
  missing_state: NO
}

[10:51:07][V][sensor:035]: 'House - Power consumption': Received new state 8.277003
[10:51:07][VV][sensor.filter:013]: Filter(0x3ffb928c)::input(8.277003)
[10:51:07][VV][sensor.filter:020]: Filter(0x3ffb928c)::output(496.620209) -> SENSOR
[10:51:07][D][sensor:127]: 'House - Power consumption': Sending state 496.62021 W with 0 decimals of accuracy
[10:51:07][D][light:093]: 'Red' Setting:
[10:51:07][D][light:102]:   State: ON
[10:51:07][D][light:125]:   Flash Length: 0.4s
[10:51:07][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 2274151077
  state: 496.62
  missing_state: NO
}
[10:51:07][V][sensor:035]: 'House - Total energy': Received new state 367.000000
[10:51:07][VV][sensor.filter:013]: Filter(0x3ffb9444)::input(367.000000)
[10:51:07][VV][sensor.filter:020]: Filter(0x3ffb9444)::output(0.367000) -> SENSOR
[10:51:07][D][sensor:127]: 'House - Total energy': Sending state 0.36700 kWh with 3 decimals of accuracy
[10:51:07][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 2690257735
  state: 0.367
  missing_state: NO`

Fossdal, 2021-08-11:
I have the same issue.
i have a 357lf sensor. and it is gaffa taped to the meter, and the meter is inside a closet. so no external light.
i have tried to test it outside. when i cover the sensor it gives a reading. and when i uncover it again it gives a reading again. so it almost looks like it is giving a reading on both edges (up and down)

WizBangCrash, 2021-08-16:
I have the same issue too. I and using an esp8266 too. I have tried adjusting the light sensor sensitivity etc. and my meter is in a dark cupboard, so no additional light to confuse the sensor.
Could this be esp8266 related, as the original suggests using an esp32?

WizBangCrash, 2021-08-16:
I think I may have root caused this one.
I read about the `internal_filter` option for the pulse meter sensor on esphome.io and added the following line to my configuration file:
```
    internal_filter: 50ms
```
Since adding this line I have had no false readings for the last hour and they used to occur every few seconds.
The default value for `internal_filter` Is 15us, so I suspect the meter LED is registering as multiple flashes since it is probably driven from a PWM :-)

Fossdal, 2021-08-16:
I have tried with the internal_filter with different values, but no different behavior. i also tried with at pull down resistor. 
when i tried with a 100nF capacitor it worked. so it has to be some kind of noise

PhantoMMM, 2021-08-19:
I also solved my problem using a capacitor.

klaasnicolaas, 2021-08-19:
In that case, I think we can close this issue?

PhantoMMM, 2021-08-19:
Yes, I think you can close it. Thanks !

closed 2021-08-19 by klaasnicolaas

moorep999, 2021-09-18:
Since I’m new to electronics, would someone please ELI5 where this 100nf capacitor sits. I’m having a nightmare getting accurate readings.