klaasnicolaas/home-assistant-glow #16: Extra Config for LDRs

maverikgrass, 2021-08-09:
I know that using an LDR isn't recommended, but for some reason finding a 4 pin photodiode or similar is impossible down here in Australia. The only way I can get one is from Alibaba or similar and I don't want to wait that long. Can anyone share with me the extra config and different wiring guide for using an LDR thanks!

jesserockz, 2021-08-09:
You can attempt to use an LDR in the same way as a photodiode. But you may have issues with the voltage not getting high enough to trigger the pulse_meter to get a reading or false readings. You will need to experiment with it.

jesserockz, 2021-08-09:
Also see #10

maverikgrass, 2021-08-09:
Thanks, I'm a little bit of a noob when it comes to electronics, is there anything else I need to add, a resistor or something? And is there a specific GPIO pin that would be best to get used. Also this is first time using ESPHome or anything or the sort, is there any code I have to edit/add?
Side note I'm using an esp8266 instead. I got the webgui and everything working, I just can't figure out the photoresistor.

TomW1605, 2021-08-09:
would a photo transistor like this one work better than an LDR?
https://core-electronics.com.au/photo-transistor-light-sensor.html (aussie site)

maverikgrass, 2021-08-09:
Honestly I have no clue, like I said, I'm an electronics noob. This is essentially my starter project.

maverikgrass, 2021-08-09:
I need some help here, I'm stuck. I've got all the pins hooked up. but It just doesn't do anything.

ArkhonTiger, 2021-08-09:
Ahoy!  I am Australian too and also had issues getting the right photodiode. I tried a few from Jaycar but they were all tuned for Infrared and wouldn't pick up the tiny light on my meter.

I also had no luck with _any_ LDR. In the end, I got a working solution with this:

Get this 4-wire board from Jaycar:
https://www.jaycar.com.au/arduino-compatible-microphone-sound-sensor-module/p/XC4438

Remove the microphone and in its place solder on a photodiode from this thing:
https://www.jaycar.com.au/photo-interrupter/p/ZD1901  (you'll need to break it open to get at them.

 Its a hacky job but its working perfectly now.

maverikgrass, 2021-08-09:
Thanks! I've ordered those two bits for pick up tomorrow and I'll give It a shot. Any tips on hacking apart the photo interrupter?

benbeton, 2021-08-09:
will it work with this one: would be a nice an cheap solution, haven't found the listed part in germany
https://www.amazon.de/gp/product/B00EFOTJZE/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

ArkhonTiger, 2021-08-09:
> Thanks! I've ordered those two bits for pick up tomorrow and I'll give It a shot. Any tips on hacking apart the photo interrupter?

I just used pliers and side cutters. It's not a particularly solid thing, just take it slow so you don't mangle the legs.

You'll eventually see a small tab thats holding them in place. Pry/cut the tab and you'll be good to slide it out.

jesserockz, 2021-08-09:
> Get this 4-wire board from Jaycar:
> https://www.jaycar.com.au/arduino-compatible-microphone-sound-sensor-module/p/XC4438

Yeah these boards underneath are very common for converting analog to digital with a user changeable threshold.

maverikgrass, 2021-08-09:
Just a side note, I've completely isolated the issue to being the Photoresistor, I simply tested by shorting the to leads of the LDR and it registered as a pulse so its not a code issue, Phew! I'll update you all tomorrow once I test the new sensors.

maverikgrass, 2021-08-10:
Is there a specific direction I have to wire the component I ripped out? The longer leg to the right or?

maverikgrass, 2021-08-10:
Well I concede I do not have the skills nor the patience to try and solder this tiny piece to the board, I've completely burned up the pads on the board, and now I'm just going to have to wait 8 months for the original board to get here from china.

closed 2021-08-10 by maverikgrass

walkerleigh, 2021-08-10:
I was going to say that the emitter should be connected to the trace which runs down to the 'G' terminal

maverikgrass, 2021-08-10:
Oh, thanks for your help anyway. To be honest I kinda lost my cool there. Not to worry though, I got a regular old LDR working. I know they're worse, for quite a few reasons, but this isnt mission critical for me so it'll do.