klaasnicolaas/home-assistant-glow #17: Every other power consumption reading is 0W.

bgmn, 2021-08-09:
Every other power consumption reading is 0 W. I can see that the photodiode works correctly from its led and it flashes just like the led on the power meter. I have a meter that has 10000 impulses per 1 kWh.

I have shortened the config because i have no need for the leds or web server.

`---
#
substitutions:
  device_name: home_assistant_glow
  friendly_name: House
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO12

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home_assistant_glow"
    version: "1.1.2"
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "redblock"
  password: "si(ni2neb3fanhGEe!vo4neGna"
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${device_name}'
    password: "luVL44uRRwpW"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "asidna3tq3t9ah2A)RF#HA9ah3fa"

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

sensor:
  - platform: pulse_meter
    internal_filter: 8us
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    filters:
      # multiply value = (60 / imp value) * 1000
      - multiply: 6
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        - multiply: 0.0001`

klaasnicolaas, 2021-08-09:
I think you have to play with the `internal_filter` a bit, because you get 10x as many pulses for 1 kWh there is a good chance that it doesn't count because it is so many times in a row 🤔

bgmn, 2021-08-09:
I have tried it with 4 and 8 us, but it doesn't change anything. My consumption is under 1000 Watts so the led isn't blinking too fast.

Here is a portion of my log
```
17:26:47][D][sensor:127]: 'House - Power consumption': Sending state 960.00000 W with 0 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Total energy': Sending state 5.67330 kWh with 3 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Total energy': Sending state 5.67470 kWh with 3 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Power consumption': Sending state 962.56689 W with 0 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Total energy': Sending state 5.67480 kWh with 3 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Total energy': Sending state 5.67580 kWh with 3 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Power consumption': Sending state 962.56689 W with 0 decimals of accuracy
[17:26:47][D][sensor:127]: 'House - Total energy': Sending state 5.67710 kWh with 3 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Power consumption': Sending state 952.38098 W with 0 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Total energy': Sending state 5.67730 kWh with 3 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Total energy': Sending state 5.67830 kWh with 3 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Total energy': Sending state 5.67850 kWh with 3 decimals of accuracy
[17:26:48][D][sensor:127]: 'House - Total energy': Sending state 5.67990 kWh with 3 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Power consumption': Sending state 962.56689 W with 0 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Total energy': Sending state 5.68080 kWh with 3 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Power consumption': Sending state 954.90723 W with 0 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Total energy': Sending state 5.68090 kWh with 3 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Total energy': Sending state 5.68200 kWh with 3 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Power consumption': Sending state 960.00000 W with 0 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Total energy': Sending state 5.68250 kWh with 3 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[17:26:49][D][sensor:127]: 'House - Total energy': Sending state 5.68330 kWh with 3 decimals of accuracy
[17:26:50][D][sensor:127]: 'House - Total energy': Sending state 5.68440 kWh with 3 decimals of accuracy
[17:26:51][D][sensor:127]: 'House - Total energy': Sending state 5.68480 kWh with 3 decimals of accuracy
[17:26:51][D][sensor:127]: 'House - Total energy': Sending state 5.68590 kWh with 3 decimals of accuracy
[17:26:51][D][sensor:127]: 'House - Power consumption': Sending state 960.00000 W with 0 decimals of accuracy
[17:26:51][D][sensor:127]: 'House - Total energy': Sending state 5.68600 kWh with 3 decimals of accuracy
```

bgmn, 2021-08-09:
I think that this is not the case. If I calculated correctly my meter will flash 20 000 times per hour, if I'm using 20 kWh. That means 55,56 pulses per second. So 1 second equals 1 000 000 microseconds. One pulse will have time to rise and fall for 18 000 microseconds.

I tried also with pulse_counter but it calculates usage and consumption in certain time period. Pulse meter calculates it every time when there is a pulse. As you can see from my logs the total energy rises every time it is seen on the logs but for some reason consumption is always every second time zero.

The timeout for zero pulses per minute is 5 minutes so this can't be the case.

Is anyone else able to reproduce this error? How can I debug this

bgmn, 2021-08-09:
Still thinking about this. I looked at the logs and there are sometimes huge consumption numbers 

It baffles me that the total energy seems to accumulate correctly even though consumption is zero or 360 000 W. Could there be some kind of error with the pulse_meter library or do I have defective esp32?

```
[23:02:34][D][sensor:127]: 'House - Power consumption': Sending state 853.08057 W with 0 decimals of accuracy
[23:02:34][D][sensor:127]: 'House - Total energy': Sending state 0.07910 kWh with 3 decimals of accuracy
[23:02:34][D][sensor:127]: 'House - Power consumption': Sending state 360000.00000 W with 0 decimals of accuracy
[23:02:34][D][sensor:127]: 'House - Total energy': Sending state 0.07920 kWh with 3 decimals of accuracy
[23:02:35][D][sensor:127]: 'House - Power consumption': Sending state 0.00000 W with 0 decimals of accuracy
[23:02:35][D][sensor:127]: 'House - Total energy': Sending state 0.07980 kWh with 3 decimals of accuracy
```

LarsNorgaard, 2021-08-10:
I think you have to use ms instead of μs. They say for power-meter it could be 50-100ms.

bgmn, 2021-08-10:
> I think you have to use ms instead of μs. They say for power-meter it could be 50-100ms.

Yes! That was it. All the documentation talks about μs and I of course thought that I have to use them too. Now I have 5ms as an internal_filter value and it works.

Thank you for you help!

closed 2021-08-10 by bgmn