klaasnicolaas/home-assistant-glow #18: Active/Reactive power measurements

phrfpeixoto, 2021-08-10:
Hi,

I'm not sure this is the right place for this question, so if it's not, please feel free to close this issue/point me elsewhere.

TL-DR: I'm know very little about power measurement: Do I need to bother about reactive power? If so, how do I measure it?

The energy meeter (3-phase, 220v AC) in my apartment has 2 LEDs, and they are labeled as "Active" and "Reactive". So far I could only notice the Active blinking.

Now I know very little about energy measurement, but I've read a bit about this matter online.
Most of the material is either too simplistic and offers naive comparisons, or is too technical and mention other terms that I also don't have a full grasp of, like the _power factor_, being _inductive_ or _capacitive_, or " reactive bounces back and forth between source and load" (whatever that means)

I do understand that Reactive power is not actually consumed by circuits, so some countries/providers do not charge for this kind of load/"consumption".
Since my meeter does mention it, I'm assuming that my provider does account for it (either direct or indirectly).
I tried [their website](https://translate.google.com/translate?hl=&sl=pt&tl=en&u=https%3A%2F%2Fservicos.cosern.com.br%2Fcomercial-industrial%2FPages%2Fenergia-reativa.aspx), but information in there is imprecise and confusing (I guess **very few** customers ask these questions), so I really don't know whether or I pay for it.

I'm having a hard time grasping:
- Whether or not I should bother/care about reactive power measurement
- How do I measure reactive power
- Whether or not reading only the active power will be representative of my actual energy usage/footprint
- Understand the core concepts

I'd appreciate if anyone could help me get tho these points, or point me in the right direction.

xorbital, 2021-08-10:
The website you've linked doesn't work for me, however it contains "commercial-industrial", so I'm not sure it applies to you.
Most private consumers won't have to pay for that (directly). Your meter only has one reading on the screen, right?
Reactive Power is the energy lost in transfer. That might be just line resistance (all the hundreds of meters the current needs to travel) or losses in inductive / capacitive circuits. If you want to learn a bit more, I'd recommend watching [this video](https://youtu.be/J86QK0Njfv4) about scam products claiming to reduce your energy bill.
So,

- I don't think you should bother, you could call your energy provider to confirm.
- You could set up two ESP sensors. But maybe the meter isn't even monitoring reactive power.
For an end device you can hook up a power meter, some can read out the reactive power / power factor.
- It'll be good enough. There's line resistance between the street and your house connection, and that doesn't get taken into account, neither does the wiring from your house to the power plant, although you're using 0.001% of it.
There are a lot of losses on the entire grid and nothing works with 100% efficiency, neither power plants, nor wiring, transformation stations nor end devices.
For me anything other than the active power is out of scope.

phrfpeixoto, 2021-08-10:
@xorbital Thank you so much for taking the time to reply here. Electroboom is the best!

That's cool, but I think I'll have to call the provider and try to find the exact info (god help me with that):
I found the [user's manual](https://www.dropbox.com/s/mk1bojvsu1eu8ne/Manual-VECTOR-PA-e-PAR-Nansen.pdf) for [my meter](https://uc25a7403714edc6248a52ddabef.previews.dropboxusercontent.com/p/thumb/ABTHmZUr-U-U77AxJOdnmOcghE8HykAJPce6TWEYS9pb96OSlaIHET_dCFnxlyvgTStFEMtOkreh_fIPqbsF2CYQTFb_2D82ySvm5Izdg-G9yjMpS3uvwIepTecqIZAEr8tuava6V0Gc_Ell0o7ogpbaxadg3awvoNfyMxDhi3jwduhnMwN9TpzG8TTeUXJHMX1dlc5H9wACUOdTJXUMgtiaQcKEpvFuA_n-8OOIZxBVuK9CzjIRIts7DQeEuJ55JwKyrdykrz77aePXemuhxPxqmgIGuUXrASIdpMEhG_Tlpt5nW9epnE_N0sg2RoQ5N5mADvaXIi5QSGbob9hDSbdtRb9DZyevzCXoceV8z16R3_WI2UAMpjFZdrOLNTW_UjElY0GPKoGeFReGTQ0LINuc/p.jpeg?fv_content=true&size_mode=5), and everything leads me to believe that some (inductive) reactive power is being measured:

- The faceplate has an indication lead for reactive power, as well as indications on how much energy each pulse accounts for.
- The manual states "To check the accuracy of the device use the LED's or the pulse outputs (KYZ) that emit, all the time, pulses proportional to the measured active and reactive energy."
- The manual contains a section with differential equations for how everything is calculated, and it does include one for Reactive power
- The display information changes every 6s, and among other info, it displays it shows Power Factor and "Total Inductive Power".

Now, all this means that the device is able to measure/monitor reactive power. It does not mean that the provider is actually charging for it. Hence, I have to call them.

The most frustrating thing is that this device is equipped with an async RS485 serial port that outputs the metering information. The manual even details how each payload is structured, but I **strongly** doubt that the provider will let me use that.

I'll go ahead and try to build a HA Glow device, get it installed, and see what measurements I get from it, and how close to the official measurement from the power provider it will be.

closed 2023-01-25 by klaasnicolaas

phrfpeixoto, 2023-01-25:
=(

klaasnicolaas, 2023-01-25:
@phrfpeixoto ??

phrfpeixoto, 2023-01-26:
Just a sad realization that I still haven't built my sensor because of lacking/misinformation.
The provider simply won't share any meaningful information and even came to lengths of saying that I could be fined if I "tamper" with my meter. Now what is the extent of tampering under their lawyers' understanding, that's a real unknown.