klaasnicolaas/home-assistant-glow #19: Calculating Remaining Units

jacauc, 2021-08-10:
I love this integration. Got it working quite well with an LDR since I don't have access to photodiode.

I have a prepaid meter where I load e.g 500 units at the beginning of the month. Since I can look on the display to see there are currently 537.1 units left, I'm looking for tips on how I can manually enter this into HA and then expose/calculate the remaining units in HA (since glow is able to tell me how many times the light blinked - i.e. how many units I have consumed.)

Any nifty ways for achieving this in HA?

fourtrax01, 2021-08-13:
I would use the monthly utility meter on your sensor data from your glow it will reset every month but will not reset your overall consumption from your glow. It will count up but with a quick calculation you should be able to get it to count down. either way you will be able to see how much you have used and it automatically resets every month.

iamvlanone, 2022-01-27:
I am doing this with a template sensor in HA. I have a input-number helper for the starting units, where I will enter the starting units and then subtract the total usage from that. I still need to decide if I want to reset my total usage everytime I enter the starting units for the month, but got it working in principle.

[image]

template sensor config:
- platform: template
  sensors:
    kwh_available:
      value_template: '{{ ((states.input_number.starting_kwh_units.state | float | round(2))) - ((states.sensor.power_total_energy.state | float | round(2))) }} '
      friendly_name: Available kWh Units
      unit_of_measurement: kWh

iamjb, 2022-02-25:
@iamvlanone 
> template sensor config:

I am trying to get this to work. I have created the input-number helper, but i am a little lost as to where the template sensor config goes. I get errors when i put it in config.yaml

Would you mind giving me a little more insight?

iamjb, 2022-03-09:
Sorry for the late reply @iamvlanone . 
Thank you for your reply. 
Now that I had some time i had a look at the log file and noticed that there was an error on the template sensor where your sensor was:
states.sensor.power_total_energy.state
Mine was:
states.sensor.house_total_energy.state

Now it seems to be working perfectly. 

Thank you!
JB