klaasnicolaas/home-assistant-glow #27: replacement for photodiode

itsboo07, 2021-08-19:
hi this is soo cool project...and here their is no availability of photodiode module .
found this module almost similar to photodiode.....could this work the same way as photodiode in this project?

https://robu.in/product/infrared-receiver-module-arduino/?gclid=CjwKCAjwgviIBhBkEiwA10D2j4302HfhiZaBzijz63ic8JA34tcEkILGdosSJzS8ykqUkMtnmlXTCRoCmz8QAvD_BwE

and also how accurate is this HA glow works?

klaasnicolaas, 2021-08-22:
Do you use the Glow?

lemppari, 2021-09-22:
This is not actually a glow issue and should be handled in home assistant. You might want to check this video https://youtu.be/UfLpVxS2QV4 I think it should give you a nice start for your problem.

closed 2023-05-07 by klaasnicolaas