klaasnicolaas/home-assistant-glow #28: LED constantly on since updating

aptonline, 2021-08-20:
Since updating the glow yaml to 1.2 and esphome 2021.8.0 the LED on my glow sensor in constantly on where as prior to this it flashed in sync with the pulse of my smart meter.

Is there a setting that needs to be changed in the yaml?

klaasnicolaas, 2021-08-20:
That way of entering the imp/kWh rate has changed in version glow yaml version 1.2.

aptonline, 2021-08-20:
Yes I know,  I’m using the default 1000 as this is what’s displayed on my meter.
[image]

aptonline, 2021-08-21:
Is this something to do with the `pulse_pin` changing from a number to the actual GPIO pin?

RamblingGeekUK, 2021-08-21:
Mine does seem to flash at all...

Userofhomeassistant, 2021-08-22:
I have also had continuously a red led without flash, but from sometimes it is also correctly flashing. It looks rather ad random.  As RamblingGeekUK, I have also noticed that there are 4 cables visible in the picture. The AO is likely the analog output and DO is the digital one. The use of both outputs looks not logical, isn't it? Maybe Klaasnicolaas can explain this.

aptonline, 2021-08-22:
@Userofhomeassistant did you upgrade from the earlier version of the glow yaml (1.0)? It all worked fine for me (flashed on the pulse reading) but since updating to 1.2 it's constantly on. I've set the pulse to 1000 as per my meter (photo above).

aptonline, 2021-08-22:
BTW I followed the guide as opposed to referring to the photo and I only used the 3 pins that were referenced.

[image]

Userofhomeassistant, 2021-08-22:
Answer to Aptonline, I use the newest version, i.e. 1.2.0.

aptonline, 2021-08-22:
> Answer to Aptonline, I use the newest version, i.e. 1.2.0.

But did you use the version prior to that? For me the led flashed correctly with the version prior. Also out of interest what version of esphome are you using? I’m trying to determine if this new behaviour is a change in the yaml for glow or a change that was introduced in the latest esphome.

RamblingGeekUK, 2021-08-22:
appears to be working, intermittently, it has started reading, after I confirm the sensor was working after holding my phones led to it.   I'm not getting anything plotting on the Energy￼ chart.   I'm using the 3 cables and I agree the AO if for analogue and shouldn't be needed. ￼

aptonline, 2021-08-22:
@RamblingGeekUK i’m assuming you are waiting a while to check the energy dashboard in HA? It does take a while to get the data in.

RamblingGeekUK, 2021-08-22:
It's been 11 hours... the animation was animating but that has now stopped... not sure.. was going to leave it alone today as I only installed yesterday and see what happens if anything tomorrow.

aptonline, 2021-08-22:
That seems a long time. Hopefully @klaasnicolaas will be able to assist. Maybe worth posting another issues so my LED issue doesn’t get lost 😉.

Userofhomeassistant, 2021-08-22:
In general,  I use the newest versions.  I have played around with the flash time, changed it to 150ms.  Presently, the LED works fine. However, I did not test it extensively, so no guarantee. Indeed, a little later, it was again continuously lighting. After uploading the same program without any changes,  the LED worked again fine, including blue flashing during uploading. The present conclusion is that is destabilizes after a while. But why? 

Another (additional?) reason for a continuous light might be that I have sometimes a misreading, a value far too high. This seems to be related to the sensitivity of the photodiode. If I make it a little less sensible, it does not pick up de flash of the E-meter. If I make it a bit too sensitive, it gives a continuous signal, which you can see from the green LED on the photodiode.  Finding the correct sensitivity is not easy. In the best case, the LED on the photodiode, starts to flash very rapidly while there is no flash from the E-meter. There seems to be a solution for this by placing a 100nF capacitor, as written in at another (closed) issue about instability of the reading. However, I have no clue where to add it. Unfortunately, that issue has been closed, but the applied solution with the capacitor has not been documented, the electronic scheme is missing......

aptonline, 2021-08-22:
@Userofhomeassistant I attempted to change the dial on the photodiode yesterday but didn’t see any changes. I’ll attempt to do it again today (it is a Sunday after all) and reboot the glow at each stage. As an observation the LED seems to be staying on after the first pulse is received for me, it’s almost life the led is not being reset correctly after it’s turned on or the delay is not being read correctly and therefore constant.

Sysoev86, 2021-08-22:
A similar problem. The red LED is always on

aptonline, 2021-08-22:
Adjusted photodiode sensitivity and issue still remains. There is a small led on the photodiode itself which you can use to calibrate it, too high and the led is on too low and it doesn’t fire. Adjusting this doesn’t seem to alter the LED connected to glow however.

klaasnicolaas, 2021-08-22:
@RamblingGeekUK A0 is not used, the photo shows it because this is from a test set-up. But you can just omit the 4th cable 😉

klaasnicolaas, 2021-08-22:
My Glow uses the 1.2 version and runs on ESPHome 2021.8, but I am not experiencing any problems in the measurements at the moment. As for the LED, that I'm not visually checking right now, so I'll get to that later.

@aptonline I find it strange that this has arisen after updating, what if you use an old version of the YAML config again?

aptonline, 2021-08-22:
@klaasnicolaas ok I’ve just rolled back to v1 of the yaml and it initially started working as expected (the led flashed) then after a few mins the LED stayed on like before, this seems to be different from when I use v 1.2 as it stays on after the first pulse.

I’m starting to think it’s a change in EPShome but looking at the change log I can’t see anything that would impact it.

Mincka, 2021-08-22:
It happens when there are two quick flashes. The duration of the flash is higher than the duration between the two flashes. The LED should go back to it "previous state" but the previous state is still "on", that's why the LED stays permanently "on".

Either it could be improved by using turn_on / turn_off with a short delay, or use a reset loop to force turn_off the LED every few seconds. It would not change the sensor results since this LED is just there for control.

aptonline, 2021-08-22:
Thanks @Mincka , this sounds like a solution. @klaasnicolaas can this be implemented? My esphome yaml skills aren’t the best 🤣.

Userofhomeassistant, 2021-08-23:
Thanks Mincka,

Your suggestion works fine for me. The LED is functioning no flawless.

Userofhomeassistant, 2021-08-23:
Hi Everyone,

The above-mentioned problem of my ghost readings has been solved (almost). As suggested in another post, the ghost readings might due to “noise” and the use of a 100 nF capacitor in the hardware setup was suggested......

When I applied a 10 uF capacitor, the hardware started to work fine, apart from two misreadings overnight. Maybe, I need a capacitor with a slightly higher capacitance, or somebody has another suggestion......

For those who encounter the same problem, I installed the capacitor between the 3V3 and GND pin of the photodiode. Note that the negative side of the capacitor should be connected to the GND pin.

klaasnicolaas, 2021-08-23:
I think it's fine to modify the code that calls the led flash, I tested it at home and it makes no difference to me.

aptonline, 2021-08-23:
@klaasnicolaas so on your setup on the flash interval is working as expected?

klaasnicolaas, 2021-08-23:
That's right, everything works fine for me with version 1.2 and ESPHome 2021.8 🤔 I'll create a PR to fix a few things.

RamblingGeekUK, 2021-08-23:
I just can't get mine to get consistent  readings

[image]

[image]

I think I may solder the cables, just to be sure. 🤔

klaasnicolaas, 2021-08-23:
@RamblingGeekUK Could you create a separate issue for this? To prevent situations from getting mixed up in an issue 😉

Caleroific, 2021-08-25:
> In general, I use the newest versions. I have played around with the flash time, changed it to 150ms. Presently, the LED works fine. However, I did not test it extensively, so no guarantee. Indeed, a little later, it was again continuously lighting. After uploading the same program without any changes, the LED worked again fine, including blue flashing during uploading. The present conclusion is that is destabilizes after a while. But why?
> 
> Another (additional?) reason for a continuous light might be that I have sometimes a misreading, a value far too high. This seems to be related to the sensitivity of the photodiode. If I make it a little less sensible, it does not pick up de flash of the E-meter. If I make it a bit too sensitive, it gives a continuous signal, which you can see from the green LED on the photodiode. Finding the correct sensitivity is not easy. In between, the LED on the photodiode may start to flash very rapidly while there is no flash from the E-meter. There seems to be a solution for misreadings related to electronic noise by placing a 100nF capacitor, as written in at another (closed) issue about instability of the reading. However, I have no clue where to add it. Unfortunately, that issue has been closed, but the applied solution with the capacitor has not been documented, the electronic scheme is missing......

I am having similar problems since setting up with 1.2.0 and ESPHome Version 2021.8.0. Can somebody tell me how to set the sensitivity to the correct setting. My led is either continuously on or continuously off.

closed 2021-08-30 by klaasnicolaas