klaasnicolaas/home-assistant-glow #29: Measurement on bi-directional meter (Germany/Zweirichtungszähler)

klaus1956, 2021-08-24:
Hello
My house has a bi-directional meter with an IR-Signal (10.000 imp./kWh) which works fine with the sensor in general BUT:
As this is recording exported and imported energy the results fit to other measurements of imported or exported energy quite well, but as the impulses do not have an algebraic sign (math. +/-) the signal itself does not provide sensful measurements - especially when I try to interpret the total values over time.

[image]

Does anybody have a good idea how to sepatate the imported and exported values?

[image]
Green Line: Glow Sensor - Yellow Dots: Imported - Blue Dots: Exported

yan14, 2021-08-24:
I don‘t think that this will work with this Interface. 
there is another Interface, which works with SML (https://www.msxfaq.de/sonst/bastelbude/smartmeter_d0_sml.htm).
But esphome has no library for sml, yet (https://github.com/esphome/feature-requests/issues/1041)

klaus1956, 2021-08-24:
Thank you - Interesting.. I will follow this development

ramdacxp, 2022-10-22:
The ESPHome SML component can be found here: https://esphome.io/components/sml.html.
In the picture above the SML send/receiver-diodes are hidden under the round sticker.

closed 2024-07-31 by klaasnicolaas