klaasnicolaas/home-assistant-glow #33: Values reported by Glow much higher

Joeboyc2, 2021-08-28:
Hi All,

I've recently setup the glow on my electricity meter and all seemed to be working well.
However, I have noticed that the power consumption readings are going up much quicker than I would expect.

My meter runs at a rate of 4000 blinks per kWh, and i have adjusted my config to that, but when looking in the logs i can see it is tracking each of these blinks and pushing the number up pretty quickly, rather then small increments.

Here is some output from my logs:
[09:48:15][D][sensor:131]: 'House - Power consumption': Sending state 320.39871 W with 0 decimals of accuracy
[09:48:15][D][sensor:131]: 'House - Total energy': Sending state 0.14925 kWh with 3 decimals of accuracy
[09:48:18][D][sensor:131]: 'House - Total energy': Sending state 0.14950 kWh with 3 decimals of accuracy
[09:48:20][D][sensor:131]: 'House - Power consumption': Sending state 319.03583 W with 0 decimals of accuracy
[09:48:20][D][sensor:131]: 'House - Total energy': Sending state 0.14975 kWh with 3 decimals of accuracy
[09:48:23][D][sensor:131]: 'House - Power consumption': Sending state 320.39871 W with 0 decimals of accuracy
[09:48:23][D][sensor:131]: 'House - Total energy': Sending state 0.15000 kWh with 3 decimals of accuracy
[09:48:26][D][sensor:131]: 'House - Power consumption': Sending state 321.77332 W with 0 decimals of accuracy
[09:48:26][D][sensor:131]: 'House - Total energy': Sending state 0.15025 kWh with 3 decimals of accuracy
[09:48:29][D][sensor:131]: 'House - Power consumption': Sending state 321.65833 W with 0 decimals of accuracy
[09:48:29][D][sensor:131]: 'House - Total energy': Sending state 0.15050 kWh with 3 decimals of accuracy
[09:48:32][D][sensor:131]: 'House - Power consumption': Sending state 320.51282 W with 0 decimals of accuracy
[09:48:32][D][sensor:131]: 'House - Total energy': Sending state 0.15075 kWh with 3 decimals of accuracy
[09:48:34][D][sensor:131]: 'House - Power consumption': Sending state 322.69629 W with 0 decimals of accuracy
[09:48:34][D][sensor:131]: 'House - Total energy': Sending state 0.15100 kWh with 3 decimals of accuracy
[09:48:37][D][sensor:131]: 'House - Power consumption': Sending state 320.51282 W with 0 decimals of accuracy
[09:48:37][D][sensor:131]: 'House - Total energy': Sending state 0.15125 kWh with 3 decimals of accuracy
[09:48:40][D][sensor:131]: 'House - Power consumption': Sending state 321.77332 W with 0 decimals of accuracy
[09:48:40][D][sensor:131]: 'House - Total energy': Sending state 0.15150 kWh with 3 decimals of accuracy
[09:48:43][D][sensor:131]: 'House - Power consumption': Sending state 321.65833 W with 0 decimals of accuracy
[09:48:43][D][sensor:131]: 'House - Total energy': Sending state 0.15175 kWh with 3 decimals of accuracy
[09:48:46][D][sensor:131]: 'House - Power consumption': Sending state 320.51282 W with 0 decimals of accuracy
[09:48:46][D][sensor:131]: 'House - Total energy': Sending state 0.15200 kWh with 3 decimals of accuracy
[09:48:48][D][sensor:131]: 'House - Power consumption': Sending state 320.28470 W with 0 decimals of accuracy
[09:48:48][D][sensor:131]: 'House - Total energy': Sending state 0.15225 kWh with 3 decimals of accuracy
[09:48:51][D][sensor:131]: 'House - Power consumption': Sending state 320.39871 W with 0 decimals of accuracy
[09:48:51][D][sensor:131]: 'House - Total energy': Sending state 0.15250 kWh with 3 decimals of accuracy
[09:48:54][D][sensor:131]: 'House - Total energy': Sending state 0.15275 kWh with 3 decimals of accuracy
[09:48:57][D][sensor:131]: 'House - Total energy': Sending state 0.15300 kWh with 3 decimals of accuracy
[09:49:00][D][sensor:131]: 'House - Power consumption': Sending state 319.14893 W with 0 decimals of accuracy
[09:49:00][D][sensor:131]: 'House - Total energy': Sending state 0.15325 kWh with 3 decimals of accuracy
[09:49:02][D][sensor:131]: 'House - Power consumption': Sending state 320.39871 W with 0 decimals of accuracy
[09:49:02][D][sensor:131]: 'House - Total energy': Sending state 0.15350 kWh with 3 decimals of accuracy

so when looking at my energy usage via the energy dashboard it reports really high usage over each half hour, where as my energy company dashboard shows a much lower readings (when looking at yesterday)
[image]

I will need to wait until tomorrow to get the true usage data from my energy provider for todays usage but i will update this ticket then, to give a good example.

I think the issue here must be due to the pulses but im not 100% sure how the calculation is being done

klaasnicolaas, 2021-08-28:
The piece of logs looks fine, I think there was some hypersensitivity of the sensor at that moment. So that causes a high kWh value in your energy dashboard. Maybe make the sensor less sensitive and/or make sure that the sensor is in a dark room.

Joeboyc2, 2021-08-28:
Hey @klaasnicolaas ,
The sensor is located under my stairs in a cupboard, and I am using the 3d printed sensor housing, i can try to adjust the sensor itself using the pot, but so far it seems to track the blips correctly.
No idea what happened at 4am this morning lol, everyone in my house was asleep

klaasnicolaas, 2021-08-30:
Maybe it was a one-time thing after all and is it okay now?

Joeboyc2, 2021-08-31:
Hey Klaas,

I think there must have been an issue with the code.

I recreated the config using the current yaml on github and reflashed.

It works better now but still not 100% correct.

I'll need to look in to why the usage on the energy dashboard is still
higher then what I see on my energy account, maybe the calculation is still
a bit off

sillymoi, 2021-11-03:
Hi,
I'm using ESP8266 with the photodiode, meter is giving off 500 pulses/kWh.
I'm seeing a similar issue, Hass typically reports 30% higher consumption which led me to believe that the electricity meter is reporting kVAh and not kWh but according to the documentation it is correctly reporting kWh. Yesterday i got 2x greater value reported for the day compared to the electricity company daily use. I tried keeping an eye on the setup and I clearly see ESP8266 blinking at the same time as the meter does so it seems the number of pulses is correct, perhaps it's related to the pulse width.

One other thing - I was getting values of around 150000W earlier (now i filtered out everything > 100000 with a lambda), depending on the position of the LED it would vary between 140000-150000... might be related but not sure how to diagnose it, suggestions are welcome :)

estevez-dev, 2021-11-04:
Hi @sillymoi . That's called noise. Those photodiode boards are very noisy when switching their state. This electromagnetic noise is counted as pulses and you are getting the wrong measurements. The only way to ditch it is to not accept any pulses after one was detected. Try to uncomment this line:

```
# internal_filter: 100ms
```

I'm done with this type of electricity consumption monitoring as my meter can blink too fast to filter out any noise.

sillymoi, 2021-11-04:
Hi @estevez-dev - of course, should've thought it through.. I cut off the high frequency noise with the `if-then` lambda but still have lower frequencies left, plus I probably did it in the wrong place (power device instead of total energy device used by Hass). It made more sense when I read about the difference between `pulse_counter` and `pulse_meter`.

My meter pulses a bit slower, hopefully the 100ms filter will help.

EspressoNerd, 2022-01-19:
> Looks like I have the same or similar issue. I've added a service to reset the energy total and wrote down my electricity meter data to compare it with the glow.
> 
> ```
> api:
>   password: ********
>   services:
>     - service: reset_total
>       then:
>         - pulse_meter.set_total_pulses:
>             id: electricity_pulse_meter
>             value: 0
> ```
> 
> In 12 hours electricity meter reported 12 kWh consumed while the glow reported 48kWh. I'll continue observation to find out is it always 4 times differences or the dependence is nonlinear.
> 
> My electricity meter blinks at 3200 imp / kWh.
> 
> @Joeboyc2 , please, let me know if you found a reason of your issue. Thanks.

This code fragment is great, @estevez-dev. May I ask you how I will be able to set the "Total Energy" to a value I enter in the web page? I would like the counter to reflect the value being displayed at the electricity meter.

klaasnicolaas, 2022-01-19:
That is currently not possible, you can only reset the value. Also, the service name in the default yaml file has been changed.

I wonder how useful it is to enter the value of your electricity meter, it's a nice idea, but at the moment that value will always reset when you update the firmware or interrupt the power. That would have to be solved first if you really want to benefit from it 🤔

EspressoNerd, 2022-01-20:
Resetting the value after rebooting / firmware update to the value it had before is more in sync with the understanding of a meter as a monotonically increasing value. Therefore I think it might be a good idea to reset the counter.

cholek3, 2022-11-12:
I also have an issue with energy spikes (nearly 1.000.000W 😂) i have uncommented internal_filter: 100ms and I will see if that helps.
[image]

When it comes to applying total meter reading to total energy value, you can either update that adding meter reading to the total value lambda, when you flash the ESP (see below), but you have to be quick with that. (1 on the photo). 

Or create text input field in home assistant and input energy total energy value from energy monitor, then simply add this value to total energy value from ESP (2 on the photo). Its a work around, but seems to work. 
Hopefully i explained that right. 
[image]
[image]

niemtec, 2023-01-16:
I still face a similar issue. The readings seem to be going crazy even if the sensor itself isn't showing much activity. The LED doesn't blink almost at all, is there a way to filter out noise?

klaasnicolaas, 2023-01-16:
Sounds like the photodiode is set too sensitive 😉

niemtec, 2023-01-21:
It's strange I've lowered it as much as it goes, the LED doesn't blink at all and yet readings are off the charts.

cholek3, 2023-01-21:
@niemtec
Possibly you may have the faulty photodiode (is it soldered right way round?). But to rule out code, provide your code. Also, did you uncomment "internal_filter" as mentioned above?

niemtec, 2023-01-21:
@cholek3 I have tried this with two photodiodes and had the same results. I checked them against the photo in README and appears to be soldered on correctly. It doesn't seem to react to the energy meters blinks at all and yet measurements are off the charts.

I'm getting daily readings of about 255.273 kWh which sounds completely wrong.

```yaml
---
# Home Assistant Glow
#
# Read your electricity meter by means of the pulse LED on your 
# meter, useful if you do not have a serial port (P1).
#
substitutions:
  device_name: home-assistant-glow
  friendly_name: House
  project_version: "3.1.0"
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO26
  status_led: GPIO5

  # imp/kWh rate ⬇ #
  pulse_rate: '1000'

  # Webserver credentials ⬇ #
  web_username: 'glow'
  web_password: 'esphome'

dashboard_import:
  package_import_url: github://klaasnicolaas/home-assistant-glow/home_assistant_glow.yaml

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  project:
    name: "klaasnicolaas.home-assistant-glow"
    version: "${project_version}"
  compile_process_limit: 1

# Choose the right Platform
# esp32: https://esphome.io/components/esp32.html
# esp8266: https://esphome.io/components/esp8266.html
esp32:
  board: esp32dev

# WiFi credentials #
wifi:
  ssid: 'The Galley'       # Enter your WiFi SSID here. Example: `ssid: 'your_network_name'`
  password: 'Wellerman27'   # Enter your wifi password here. Example: `password: 'abcde123456'`

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: '${device_name}'

captive_portal:

# Enable logging
logger:

# Enable improv over serial
improv_serial:

# Enable Home Assistant API
api:
  services:
    - service: reset_total_energy
      then:
        - button.press:
            id: button_reset_total

ota:
  safe_mode: false
  reboot_timeout: 5min
  num_attempts: 5

web_server:
  port: 80
  auth:
    username: '${web_username}'
    password: '${web_password}'

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO4
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

button:
  - platform: restart
    name: "Restart - Glow"
  - platform: template
    name: "Reset - Total Energy"
    id: button_reset_total
    on_press:
      - pulse_meter.set_total_pulses:
          id: sensor_energy_pulse_meter
          value: 0

# Sensors for ESP version and WIFI information
text_sensor:
  # Installed version
  - platform: template
    name: "Glow - Installed version"
    id: glow_version
    icon: "mdi:label-outline"
    entity_category: diagnostic
    lambda: |-
      return {"${project_version}"};
    update_interval: 6h
  - platform: version
    hide_timestamp: true
    name: '${friendly_name} - ESPHome Version'
  - platform: wifi_info
    ip_address:
      name: '${friendly_name} - IP Address'
      icon: mdi:wifi
    ssid:
      name: '${friendly_name} - Connected SSID'
      icon: mdi:wifi-strength-2

sensor:
  # WiFi signal
  - platform: wifi_signal
    name: "${friendly_name} - WiFi Signal"
    update_interval: 120s

  # Pulse meter
  - platform: pulse_meter
    name: '${friendly_name} - Power Consumption in kW'
    id: sensor_energy_pulse_meter
    unit_of_measurement: 'kW'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    internal_filter: 100ms
    internal_filter_mode: 'PULSE'
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * (60.0 / ${pulse_rate});
    total:
      name: '${friendly_name} - Total Energy'
      id: sensor_total_energy
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 2
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});
  # Total day useage
  - platform: total_daily_energy
    name: '${friendly_name} - Daily Energy'
    id: sensor_total_daily_energy
    power_id: sensor_energy_pulse_meter
    unit_of_measurement: 'kWh'
    icon: mdi:circle-slice-3
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 2
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001

# Enable time component to reset energy at midnight
# https://esphome.io/components/time.html#home-assistant-time-source
time:
  - platform: homeassistant
    id: homeassistant_time
```

yarpsa, 2023-10-05:
I have project_version: "4.0.0" and my meter also blinks at 4000/kWh but I don't know where or how to change the setting from 1000 to 4000. Can you help me as a beginner, please?

sparkyws, 2023-12-07:
> I have project_version: "4.0.0" and I keep getting 270,000 - 285,000kWh spikes in House - Power Consumption

Change to GPIO26 and see if that fixes the issue

[https://github.com/klaasnicolaas/home-assistant-glow/issues/300](url)

stackner, 2023-12-09:
Same issue here. anyone had any luck?

tenacioustechie, 2024-07-01:
I'm having this issue too... 

- I haven't found any guidance on calibrating the sensor
- I can't find where this line of code might help `# internal_filter: 100ms` if it still might help
- my sensors is in the black 3d printed case inside a metal cabinet outside, with very little external light
- I'm using ESP32 dev board v1, I don't think the GPIO pins are an issue but will check that

The 'W' power consumption value seems correct. 
Its the 2 Total values it provides that seem inordinately higher than they should be

klaasnicolaas, 2024-07-01:
> - I haven't found any guidance on calibrating the sensor
> - I can't find where this line of code might help `# internal_filter: 100ms` if it still might help

There are no instructions yet, but it could certainly be added to the website. This also applies to the use of the `internal_filter`, information about this can mainly be found in the issues. 

./Klaas

getninjaN, 2024-07-13:
My Glow has started acting up.
Updated to the latest version through the website after a Home Assistant update which renamed something, or something, and now the consumption has gone haywire.
Any ideas?
All the components are the recommended once in the README here on GitHub.

I tried changing the `internal_filter` setting but not sure if that made a difference.

https://github.com/user-attachments/assets/4e18779f-51f4-46cf-9761-e67c127f4d91

klaasnicolaas, 2024-07-13:
Please create a new issue for this

./Klaas

Footz355, 2024-08-07:
Where after installing and setting up ESPHOME GLOW can I fing config yamls? My ESP Home dashboard doesn't see my GLOW instance, HA sees it, but I have crazy Power values, and want to enable internal filter, so again, where can I find config to change the internal filter. It could be enabled on the Glow web server.

klaasnicolaas, 2024-08-07:
@Footz355 You can change the YAML config after adopting in ESPHome Dashboard. If it's not shown there you could try removing the glow from Home Assistant, then reboot to see if both HA and ESPHome see the device.

Have you already tried adjusting the sensitivity of the sensor using the small potentiometer?

Related doc: https://glow-energy.io/docs/advanced/firmware_customization

./Klaas

Footz355, 2024-08-07:
Greetings Klass. No, I have not played with the potentiometer because I am currently using Glow 3.0.0 and it is working fine with internal filter set to 200. I just got lost in new config style with separate sub configs. Probably will have to download it, edit in internal_filter, copy/paste it to a single yaml and deploy in my ESPHome, at least that seems the easiest way for me. Or I will just stay with 3.0.0 until some ESPHome update will brake it. My I recommend adding internal_filter setting to web_server? It seems judging by my experience and browsing in forums that this feature is more often used then changing pulse rate settings from the glow web_server. Thank you very much for this project.

closed 2025-07-12 by Joeboyc2