klaasnicolaas/home-assistant-glow #34: Photodiode sensitivity not enough for led detection + red led stays on

nygellyndley, 2021-08-28:
Hi, I'm having a couple of issues :

1. I have the recommended photodiode but doesn't seem to be sensitive enough to pick up the meter led
2. I can use a torch to trigger the photodiode but often the red light will tend to stay on permanently

On (1) I've adjusted to pot so the `D0-LED` on the photodiode board is just off.  From there I can get the red led to light and see a detection registering in the logs when I get a torch and shine it right up to the photodiode.

It's entirely possible this is some hardware fault with the photodiode sensor, or that this version just isn't sensitive enough or something, but I thought I'd ask since it's not hugely clear to me.  Given the ESPHome sensor type is essentially "pulse" I suspect all the logic around outputting a pulse to D0 is done on the photodiode board, so this is probably not some code or config issue?  Maybe I should reorder the same board or perhaps there's an alternative with better sensitivity?

On (2) I can use a torch to trigger the photodiode and I see log entries appearing for the detection.  Moving the torch away the red led stays on though, my understanding is it should turn itself off? 

Worth noting that even though the led stays on, moving the torch back over the photodiode does register a secondary detection, so if this is an issue it doesn't seem to be with light detection as such, more around turning on/off the led.

klaasnicolaas, 2021-08-28:
With the debug led something will change, because there are more people who have the same problem that the led stays on.

adam-80, 2021-08-29:
I am experiencing the same issue, I have spent hours tuning the photodiode sensor and just can't get it to detect the LED on my meter. As above, I was able to simulate the meter with a torch and it worked, so the setup from a wiring and YAML point of view is working correctly. 

I am think I will order another photodiode sensor as it seems to be the sensor not being sensitive enough. I wonder if anyone else has had this issue?

For clarity, I am using a Landis+Gyr e360 smart meter, it has a red pulse LED and 1/Wh = 1000/kWh.

Any ideas on what else to try whilst I wait forever for my new sensor?

nygellyndley, 2021-08-29:
Landis Gyr + E470 here.  But even though you could say it isn't the brightest LED in the world I do feel this probably isn't a smart meter issue as such.

For example given the red led issue staying on issue above, I tried using the Glow's own LED (which is much brighter) to trigger the diode and I couldn't even get that to work, only a bright torch shone right into the center of the diode right up next to it seemed to work (even after trying to calibrate via the pot), to me that seems like it's just the diode unit itself isn't really suitable.

adam-80, 2021-08-29:
I agree it seems to be the sensor/diode itself that's the issue. I am going to order a couple more to test, at least they're cheap. 

Your idea of trying a different diode is an interesting one, I am not sure if the IC would be tuned to this one in particular, but worth a try if new ones don't work.

cegenolf, 2021-08-29:
Please check whether the photodiode is the right way around, [as shown in the picture](https://www.sunrom.com/p/light-sensor-module-photodiode).  My two modules that I ordered were the wrong way round.  The cathode must be connected to "+" not to "-".  After turning the photodiode over, it worked with low sensitivity on the poti.  But I also have the continuous glow of the red LED, since Esphome 2021.8.x, before with 1.20.4 it was okay.
Hope that helps...

adam-80, 2021-08-29:
Oh... My... God.... It is soldered on backwards. Thanks for the suggestion! I will flip and resolder and see how it goes.

nygellyndley, 2021-08-29:
> Please check whether the photodiode is the right way around, [as shown in the picture](https://www.sunrom.com/p/light-sensor-module-photodiode). My two modules that I ordered were the wrong way [...]

This image is showing it the wrong way round or the right way round?  Mine is soldered the opposite way, is that correct cathode should be negative right?

cegenolf, 2021-08-29:
https://www.analog.com/en/technical-articles/optimizing-precision-photodiode-sensor-circuit-design.html

I found this a few das ago and i tried to flip the Photodiode, then it works well. 
The Image is showing the right way round. It is Not a LED!

KixMan28, 2021-08-29:
> Please check whether the photodiode is the right way around, [as shown in the picture](https://www.sunrom.com/p/light-sensor-module-photodiode). My two modules that I ordered were the wrong way [...]

I think the picture in the link is the wrong way. Mine is soldered the other way. I googled this topic and everywhere  I found that the LED flat side is the negative (cathode). I have checked my module and the flat side is soldered to the negative "-" side on the board . Does this mean that the printing on the module is wrong? I checked my module with multimeter and the GND and the "-" leg on the diode has continuity, So I am confused now.

cegenolf, 2021-08-29:
IT is NOT a LED, see Post before...

nygellyndley, 2021-08-29:
> 
> 
> IT is NOT a LED, see Post before...

Gotta say, awesome work spotting that.  Thanks that's saved so much messing about.

KixMan28, 2021-08-29:
> 
> 
> IT is NOT a LED, see Post before...

Thank you. I haven't seen your previous reply, as I was writing my comment and  the page didn't refreshed before I have submitted my comment. I ordered mine from Banggood so it is probably the same batch as @nygellyndley has. I will turn around my photodiode immediately.

cegenolf, 2021-08-29:
I bought mine on ebay.de, shipped from China.  The pick and place robot probably also believed it was an LED 😂

Nice that I could help, I read and write most of the things with Google translator because my school English is very rusty 🙄

Eddie46848, 2021-08-29:
> Please check whether the photodiode is the right way around, [as shown in the picture](https://www.sunrom.com/p/light-sensor-module-photodiode). My two modules that I ordered were the wrong way [...]

Thanks for this, was messing with mine for hours. 2 diodes sent the wrong way round. Desoldered and it was working as it should. Thanks

KixMan28, 2021-08-29:
I have rotated the photodiode and now it works correctly. Thank you @cegenolf  for the hint.

bedfelllow, 2021-08-30:
I've got some of photodiode modules on order. Just for clarity, should the flat of the photodiode be on the + or - marking of the PCB?

https://m.media-amazon.com/images/I/51VEP0osl1L._AC_SL1100_.jpg

KixMan28, 2021-08-30:
If you have the photodiode is mounted as in the picture on the link, than it is correct. Check the metal parts inside the photodiode. The bigger part should be "+".

adam-80, 2021-08-30:
Just wanted to come back in and say that my PD was indeed mounted backwards. Was actually a bit annoying to get de-soldered but I managed to switch it around and bang, working like a charm.

mark-10-5, 2021-09-04:
Life saver. Mine was shipped backwards also and now is working great with it the correct way around.

klaasnicolaas, 2021-09-05:
I think we should put some more in the docs on this, as this seems to be a more common problem 🤔

KixMan28, 2021-09-05:
@klaasnicolaas Hi, as many of us encountered the issue of the wrongly soldered photodiode, would you be so kind to mention this in the readme, so everyone is aware of this, and they don't start spaming the issue section with this topic.

klaasnicolaas, 2021-09-05:
That was exactly what I suggested 🤣 Does anyone have pictures of a photodiode connected right and wrong way?

KixMan28, 2021-09-05:
@klaasnicolaas Here are some pictures which are with the correct orientation:
https://www.sunrom.com/p/light-sensor-module-photodiode
https://m.media-amazon.com/images/I/51VEP0osl1L._AC_SL1100_.jpg

... and here is my module with the wrongly soldered photodiode.
[image]

bruvv, 2021-09-05:
And indeed I also had this issue... very handy to update the readme to show this!  (thanks @KixMan28 that was very clear!)

nygellyndley, 2021-09-05:
> 
> 
> And indeed I also had this issue... very handy to update the readme to show this! (thanks @KixMan28 that was very clear!)

Created a pull request : Created a pull request for this : https://github.com/klaasnicolaas/home-assistant-glow/pulls

@KixMan28 I used your image I hope you don't mind!

KixMan28, 2021-09-05:
@nygellyndley no problem.

pschneider87, 2021-09-06:
just a quick heads up before I go and unsolder the diode:

when changing the potentiometer, I can make the diode recognize the LED in the meter, at least it's showing something on the own status LED, and the red LED in the case is flashing briefly

but my diode looks exaktly like the one in https://github.com/klaasnicolaas/home-assistant-glow/issues/34#issuecomment-913155023

so I'm wondering what is going on here...

bruvv, 2021-09-06:
> 
> 
> just a quick heads up before I go and unsolder the diode:
> 
> when changing the potentiometer, I can make the diode recognize the LED in the meter, at least it's showing something on the own status LED, and the red LED in the case is flashing briefly
> 
> but my diode looks exaktly like the one in [#34 (comment)](https://github.com/klaasnicolaas/home-assistant-glow/issues/34#issuecomment-913155023)
> 
> so I'm wondering what is going on here...

I had the same, when changing the sensitivity dial, it did pick up the led. But it didn't report the correct values. Often my values were around 72000 watt although it was actually around 100watt... after flipping the diode it reported it correctly.

pschneider87, 2021-09-06:
It's strange. I have desoldered it and resoldered in correct way.
I can drop the sensitivity now to make it recognizing the IR-LED, but the values sent to Home Assistant are still not correct: makes about 2 kW/h per minute :-o

pschneider87, 2021-09-06:
I played around with the potentiometer, and made it close to the maximum before it doesn'T stop recognizing anymore.
Now I get a more realstic measurement, however, it seems that this is not updating even if I'm starting water heater, coffeemaschine and the iron.
It's always around 358-362 Watts

bruvv, 2021-09-06:
@pschneider87 please edit your post instead of double posting to keep the overview. Also I would say double check that your diode is getting all the flashes, especially if it is flashing raptly. With all that stuff on I would imagine it will be a disco light on your electricity meter. also make sure it isn't exposed to other light so make sure you block any light from exposing the diode.
Besides this also make sure you have adjusted your pulse_rate

pschneider87, 2021-09-06:
@bruvv sorry for double posting, will remember this.

Regarding the issue:
the Meter  LED pulses 61 times per Minute, Pulserate in the script is set to 10000. HA Glow exposing 358 to 362 Watt which is correct for this pulse rate.
Checking the internal power consumption of the meter returns ~ 250 Watt, which should be ~ 42 Pulses per Minute
So maybe the meter is just not as correct as it should be 

Maybe someone else with an Itron 3 HZ meter can confirm if this is an generall issue with this meter, or a specific for mine...

bedfelllow, 2021-09-08:
Just received my modules, which also have the photodiode wrongly orientated. 

[image]

el97, 2021-09-08:
So I am not the only one with this problem then... I have some basic soldering skills. Is it easy to fix on my own?

bruvv, 2021-09-09:
> So I am not the only one with this problem then... I have some basic soldering skills. Is it easy to fix on my own?

It's very very easy. Otherwise send it back or get a refund.

bedfelllow, 2021-09-09:
> So I am not the only one with this problem then... I have some basic soldering skills. Is it easy to fix on my own?

Took me about 5 minutes with just a soldering iron and some solder. Potential you could lift a track though if you are not proficient with soldering.

cliffordwhansen, 2021-09-09:
I also got one from Banggood that was the wrong way round, after fixing that it now detects the pulses from a torch, but its not detecting the pulses from my meter, I have even adjusted the sensitivity to just after it stops flashing like crazy on its own.

I even tried covering the diode so no ambient light gets in.

pschneider87, 2021-09-09:
> I also got one from Banggood that was the wrong way round, after fixing that it now detects the pulses from a torch, but its not detecting the pulses from my meter, I have even adjusted the [...]

Did you properly sold the ESP Pins? I had some pins not being connected well, with the same experience

cliffordwhansen, 2021-09-09:
> Did you properly sold the ESP Pins? I had some pins not being connected well, with the same experience

I'm not seeing the D0 light on the board light up when the meter flashes, but it does when the torch flashes.

I will resolder them just in case (also I have daisy chained the dupont cables from the esp32 to the photodiode board (will remove one link to test)

slovdahl, 2021-09-10:
Oh wow, mine also seems to be from a batch with the diode the wrong way. Thanks so much!

klaasnicolaas, 2021-09-15:
It took a while but I've updated the documentation with better illustration and a separate section at the bottom as FAQ.

nygellyndley, 2021-09-16:
There's still the red led staying on but I think that's already logged elsewhere, happy to close this

klaasnicolaas, 2021-09-16:
Code has already been changed for the red light, so that should be solved. For now I'm closing the issue.

closed 2021-09-16 by klaasnicolaas

wiesieks1, 2021-11-30:
where should I get the firmware for ESP32, the link at the bottom of the page is blocking me and not letting me download anything
this (<script type = "module" src = "https://unpkg.com/esp-web-tools@5.2.0/dist/web/install-button.js?module"> </script>)

wiesieks1, 2021-11-30:
is there a tutorial on how and what to upload to ESP 32, I'm new, stupid and I don't know what's going on, I can't find it

klaasnicolaas, 2021-11-30:
This issue has already been closed, which means that you will not respond with new issues that are not related to the first post.

Please open a new issue with an explanation of your problem.

wiesieks1, 2021-11-30:
can't you just help me get these files for ESP and home assistant to build a sensor like this?

klaasnicolaas, 2021-11-30:
You can use the website to (via improv) put the firmware on your ESP or you can download the yaml file, like the instructions under get started. For further text and explanation, please create a new issue, since your request for help has nothing to do with this issue.