klaasnicolaas/home-assistant-glow #37: Is anyone using an ESP8266?

bedfelllow, 2021-08-30:
The project page makes reference to ESP32/8266. Does anyone have any experience using it with an ESP8266? I know they are both cheap cards, but I have spare ESP8266 cards which I try to use up for project that are suitable.

KixMan28, 2021-08-30:
Yes, I have installed it on an Geekcreit ESP-12F module. It is still in test phase, but it seems to works. Please make sure that you change the platform and board to your specific setup.

bedfelllow, 2021-08-30:
> Yes, I have installed it on an Geekcreit ESP-12F module. It is still in test phase, but it seems to works. Please make sure that you change the platform and board to your specific setup.

I think mine are ESP-12E so presumably I just need to change the platform and board to:

`  platform: ESP8266
  board: nodemcuv2`

KixMan28, 2021-08-30:
Yes, this is what I use too.

closed 2021-08-31 by bedfelllow