klaasnicolaas/home-assistant-glow #38: Proposal for an all-in-one enclosure

bedfelllow, 2021-08-31:
I flashed this to a Wemos D1 Micro last night and although I have not tested the functionality yet, it flashed OK and it should work as it's an ESP8266. Given the very small size of the Wemos D1 Mini, this led me to think that there is potential for the sensor and MCU to be in a single case, which would be quite a neat solution I think. Unfortunately I'm not skilled at 3D design, but maybe someone in the community could design something?

slovdahl, 2021-09-10:
Is yours really called Wemos D1 Micro? I can't find any references to such a board.

I use [Wemos D1 Mini](https://www.wemos.cc/en/latest/d1/d1_mini.html) and [D1 Mini Pro](https://www.wemos.cc/en/latest/d1/d1_mini_pro.html) myself, so I will at least try to come up with modified cases that fit those two at some point.

bedfelllow, 2021-09-10:
Sorry, I got the name wrong. You are correct, it is a Wemos D1 Mini. I think a case incorporating the Di Mini and the photodiode board would be a great all-in-one solution.

m-s-e, 2021-09-13:
@slovdahl how did you design the cases? If it's Fusion360 I'd be happy to help with taking the design files and creating something for the D1 Mini

slovdahl, 2021-09-13:
> @slovdahl how did you design the cases? If it's Fusion360 I'd be happy to help with taking the design files and creating something for the D1 Mini

I haven't designed anything yet :slightly_smiling_face: I'll try to modify the current ones whenever I can find the time.

Maybe @klaasnicolaas can shed some light on how they were originally designed?

m-s-e, 2021-09-13:
@slovdahl - sorry!! i thought you have designed the original enclosures!! 🤦‍♂️

KixMan28, 2021-09-13:
@m-s-e I modified a little bit the original files, because I use a Geekcreit ESP 12F instead of an ESP32 module. I also modified a photodiode case because it was too loose  for my taste.
[HA Glow.zip](https://github.com/klaasnicolaas/home-assistant-glow/files/7154929/HA.Glow.zip)

bedfelllow, 2021-09-13:
> @m-s-e I modified a little bit the original files, because I use a Geekcreit ESP 12F instead of an ESP32 module. I also modified a photodiode case because it was too loose for my taste.
> [HA Glow.zip](https://github.com/klaasnicolaas/home-assistant-glow/files/7154929/HA.Glow.zip)

Any chance you could share an STL of your photodiode enclosure modification, as I find the one I have printed to be very loose?

It would be even better though if someone was able to create the suggests all-in-one Wemos D1 Mini case.

KixMan28, 2021-09-13:
@bedfelllow see attached the 3MF files.
[Sensor case.zip](https://github.com/klaasnicolaas/home-assistant-glow/files/7155175/Sensor.case.zip)

bedfelllow, 2021-09-13:
> @bedfelllow see attached the 3MF files.
> [Sensor case.zip](https://github.com/klaasnicolaas/home-assistant-glow/files/7155175/Sensor.case.zip)

Great, thank you.

klaasnicolaas, 2021-09-13:
@m-s-e The original case is designed by me 😉
Unfortunately I can't design a case for a Wemos D1 at the moment because I'm out of stock of that board.

m-s-e, 2021-09-14:
all, any thoughts on how the boards might be arranged for an all-in-one enclosure?
on top of each other / side by side / above&below ?

bedfelllow, 2021-09-14:
> all, any thoughts on how the boards might be arranged for an all-in-one enclosure?
> on top of each other / side by side / above&below ?

I initially considered them being on top of each other as that would make a very compact assembly but I'm not sure of the practicalities of wiring the two together in this configuration. It therefore may be easier to have them both at the same level within the enclosure and I also think this would be an easier design.

bedfelllow, 2021-09-14:
A case something on the basis of this may be practical, though it would obviously be a lot shorter: https://cdn-blog.adafruit.com/uploads/2021/01/Wemos-Lolin-D1-Mini-Adafruit-dual-sensor-case-600x400.jpg

bedfelllow, 2021-09-14:
Or maybe something like this but replacing the DHT22 sensor with the photodiode board? https://www.thingiverse.com/thing:4274361

bedfelllow, 2021-09-14:
This seems to be the shorter version of the first link I posted. https://www.thingiverse.com/thing:4711281

bedfelllow, 2021-09-14:
I mashed this together using the sensor case from this project and this one from thingiverse: [https://www.thingiverse.com/thing:4304401](url)

It's obviously not ideal but as a proof of concept it works.

https://i.ibb.co/0ffM4BP/IMG-20210914-220409.jpg

https://i.ibb.co/1qvm5vc/IMG-20210914-220354.jpg

https://i.ibb.co/6FPpfbs/IMG-20210914-220330.jpg

bedfelllow, 2021-09-18:
I've combined the sensor case from this project and a modification of this case [https://www.thingiverse.com/thing:4861266/files](url), cutting the lightning bolt in the lid using this as design: [https://www.thingiverse.com/thing:2553540/files](url) to come up with this:

[image]

[image]

[image]

anverx, 2021-09-18:
This project really should live in one small box.  Here is how i wired the photo sensor to an esp32 dev board.  I'm plannig to design a small case for this.  I'm considering a vertical rail so that sticky tape attachment is only on the top surface of the metering device.
[image]

anverx, 2021-10-18:
oh yeah, that reminds me, i've sill gotta publish mine ;)
[image]

bedfelllow, 2021-10-18:
Looking forward to trying both of these designs. Thanks.

closed 2022-01-27 by github-actions[bot]