klaasnicolaas/home-assistant-glow #39: Suggestions for updating readme

DavidREWorthington, 2021-09-01:
I've got this working now but I would have found it easier with the following pointers so these are documentation suggestions that might help others;

show how to bend the sensor to fit the 3D case ( I know this must seem obvious but I scratched my head worrying I'd got the wrong component for a bit)
mention the need to adjust the sensitivity of the photodiode to suit the meter led. I found one of the meters much darker than the other and this took several adjustments to get it working
mention that it is possible to use 2 glow sensors if your solar meter also has a pulse led and integrate this with the Energy dashboard
I found the pulse_meter behaved strangely with the solar meter. I switched to use pulse_counter and it seems to be stable if slower to update - I guess it updates at a lower frequency

I defined an additional pin (I use 13) as solar_pin

great work by the way - I now have a working HA Energy dashboard working for the cost of a few sensors!

Extract from my yaml

sensor:
  - platform: pulse_meter #counter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    on_value:
      then:
        - light.turn_on:
            id: led_red
            flash_length: 400ms
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});
  - platform: pulse_counter
    name: '${friendly_name} - Solar production'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${solar_pin}
    on_value:
      then:
        - light.turn_on:
            id: led_red
            flash_length: 400ms
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);
    total:
      name: '${friendly_name} - Total solar'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      last_reset_type: auto
      state_class: measurement
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});

ben0551, 2021-09-02:
How did you power the second one? Splitter from the 3.3v on the espboard?

DavidREWorthington, 2021-09-02:
> How did you power the second one? Splitter from the 3.3v on the espboard?

I split off the 5v output on the board. I'm powering the esp32 using a phone charger. I haven't tried off the 3.3v pin. It might also be useful to note that my meter box is outside. I've run wires from the sensors to the esp32 inside the house. It seems that the circuit still works with quite long sensor wires (about 3m) and this might suit other situations.

ben0551, 2021-09-02:
Ahhhh so. I have a 30pin with only a 3.3v. So I'm unning 2 boards for on/off peak.

fourtrax01, 2021-09-05:
@DavidREWorthington  
I as well had to change to the pulse counter to get mine to be accurate. The default refresh rate of the pulse counter is 60s so it is slower to update however you can increase the refresh rate by adding a different update_interval:
I used 10s

sensor:
  - platform: pulse_counter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    update_interval: 10s
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    on_value:

klaasnicolaas, 2021-09-05:
For the glow I deliberately did not choose the `pulse_counter` component. This is not accurate at all for me, because it makes fixed steps in watts.

While the `pulse_meter` component is just a mix of the `pulse_counter` and `` pulse_width`, which should result in correct better and more accurate calculations. https://github.com/esphome/esphome/pull/1434#issue-545850359

klaasnicolaas, 2021-09-05:
As for modifying the documentation, anyone is free to create a pull request to add or modify additional information. I like the first 2 points to add more information about it.

closed 2022-01-27 by github-actions[bot]