klaasnicolaas/home-assistant-glow #40: sensor has gone crazy - sending reading even though light diode is not connected.

jurgenweber, 2021-09-01:
Hi

Yesterday I finally got my case and I set this up against my power meter. I start getting data, the data seems reasonable, all good.

This morning I have a look and the thing has gone wild! It is taking in readings at a lightning fast pace.

[image]

I thought maybe light was getting in from outside so I put some bluetac around it, still going nuts. I disconnected the light diode and readings are still coming through (thats weird :)). I have tried power cycling it, I double checked all the connections and I reflashed OTA.

```
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.69900 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.70900 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.72300 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.72500 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.73800 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.74100 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.74600 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.75900 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.76100 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.77100 kWh with 3 decimals of accuracy
[21:57:19][D][sensor:131]: 'House - Total energy': Sending state 24.78200 kWh with 3 decimals of accuracy
```
any ideas on what to do here? Or has my esp32 kicked the bucket?

klaasnicolaas, 2021-09-05:
At this point I don't really have a clue where the problem could be.

jurgenweber, 2021-09-05:
yeah, so... I think it might be hardware. I tried moving the esphome case around and it stopped spamming but now it reports nothing. If I squeeze the case it'll start spamming again. I have tried different power supplies and positions in the power box.

I am thinking I will get another esp32 and light diode and try again. Maybe I broke something. I'll report back when I get to it.

KixMan28, 2021-09-05:
@jurgenweber Hi Please check if your photodiode is soldered to the right polarity, because I had similar issues  randomly (no readind, wild reading, correct reading) until I reversed polarity. Several of us reported this issue, the photodiode has the flat side oposite to the normal LED and this might caused the wrong orientation during the priduction process. Check issue #34 for more details.

jurgenweber, 2021-09-06:
@KixMan28 Cool! I gave the thread a read. Interesting.

After pulling mine out and checking it, yes. You are correct the polarity is incorrect. Mine is the wrong way around. Mine looks exactly like the photo here: https://github.com/klaasnicolaas/home-assistant-glow/issues/34#issuecomment-913155023

I will close this thread and follow up in #34

closed 2021-09-06 by jurgenweber