klaasnicolaas/home-assistant-glow #41: ESP doesn't boot when DO connected

gcordalis, 2021-09-02:
Just received all my parts and wired it all up.

I've run into an issue where the ESP32 won't boot if DO (on the photodiode) is connected.

If I disconnect the pin it'll boot fine, I can connect to it, HA sees it - all is well minus getting no data from the sensor.

If I connect the pin after boot log output freezes with the below message:
`[D][sensor:131]: 'House - Total energy': Sending state 3.85000 kWh with 3 decimals of accuracy
[D][light:038]: 'Red' Setting:
[D][light:049]:   State: OFF`

I've replaced the wires from ESP to photodiode, resoldered the connections, run the ESP from a different power source and reflashed the firmware. I have no idea what else to try - I appreciate this is unlikely to be a glow specific issue but searching around hasn't helped.

Any help would be greatly appreciated!

landmaj, 2021-09-08:
I had the same issue. Try using some other pin, I used D3 instead of D12 and it worked.

landmaj, 2021-09-08:
According to my reseach, GPIO12 is a bootstraping pin and if pulled up it will prevent ESP32 from booting normally (the pin is used to set flash voltage or something). It is best to avoid it and use another pin if possible.

swa72, 2021-09-08:
It worked for me when I changed from 3.3V to 5V.

AndyKassell, 2021-09-12:
Same issue using ESP32 Devkit C module, I moved D0 to GPIO13 (the other side of the GND pin from GPIO 12 on this layout) and all now works as expected.

gcordalis, 2021-09-13:
Thanks everyone.

Turned out the photodiode shipped with the diode soldered to the board the wrong way. removing it and switching the pins around resolved the issue.

First time using ESP32 boards and I certainly was not expecting my first issue to be a hardware related mistake...

closed 2021-09-13 by gcordalis