klaasnicolaas/home-assistant-glow #42: Use for Gas-Meter

klaus1956, 2021-09-05:
Hello
In Germany many of the gas meters are non electric but emmit a magnetic impuls e.g 1x per 1/10 m³.
I replaced the board with the sensitive LED by a magnetic switch (KY-021 mini) and foud that it can be triggert by a magnet in the same way (like with the LED).

Cannot test it at the moment because our gas heating is not running yet - but will try to test A.S.A.P. 

May be there could be a Gas-spinn-off with m³ and names according to that new GAS function.

bedfelllow, 2021-09-06:
Sounds interesting and a good way of getting total household energy by measuring both electricity and gas. Not sure about the regulations regarding attaching a sensor to a gas meter though. Here in the UK, when Sky run a satellite cable through our gas meter box the gas authority made them remove it.

klaus1956, 2021-09-06:
In Germany there are magnetic sensors sold for the gas meters for ~70€ (the model I have) for ~70€.
So I am not too concerned (and usually the meters are only visited all 3..5 years -inbetween its self reading)

landmaj, 2021-09-14:
I just finished my sensor for gas meter so I can share my experience.

I used [this sensor from local supplier](https://botland.store/magnetic-sensors/14279-reed-switch-module-adjustable-iduino-se015-5903351242189.html), because I grew too impatient waiting for [KY-021 from AliExpress](https://pl.aliexpress.com/item/1005002742977958.html). Both should work the same.

Positioning for the sensor proved to be somewhat tricky on my meter, but the only real issue was debouncing the reed switch. ESP32 allows up to 13us on `internal_filter` in `pulse_meter` and this is not enough - I got 3-4 times more readings than I should. I found the solution [in this issue](https://github.com/esphome/issues/issues/493#issuecomment-507438088) and it worked great.

Reed switch D0 pin is connected to pin 23 on ESP32 and configured as `binary_switch`. This switch is used to generate a short pulse on pin 22, which is physically connected to pin 21. `pulse_meter` uses that pin as input, instead of directly relying on input from reed switch.

So far I got perfect readings.

Here is my config
```yaml
substitutions:
  device_name: burn
  device_description: "Measure your natural gas consumption with the magnetic pulse of your meter."
  friendly_name: Burn
  pulse_pin: GPIO23
  output_pin: GPIO22
  input_pin: GPIO21
  pulse_rate: '100' # imp/m³

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  platform: ESP32
  board: nodemcu-32s
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:

ota:
  password: !secret password

api:

text_sensor:
  - platform: version
    name: "${friendly_name} - ESPHome Version"
    hide_timestamp: true

# Connected directly to input pin used in pulse counter.
output:
  - platform: gpio
    id: gpio_output
    pin:
      number: ${output_pin}

# Used for debouncing reed switch.
binary_sensor:
  - platform: gpio
    name: "${friendly_name} - Pulse input"
    internal: true
    pin:
      number: ${pulse_pin}
      mode: INPUT_PULLUP
    filters:
      - delayed_on: 500ms
      - delayed_off: 500ms
    on_press:
      then:
        - output.turn_on: gpio_output
        - delay: 10ms
        - output.turn_off: gpio_output

sensor:
  - platform: pulse_meter
    name: '${friendly_name} - Gas consumption'
    internal: true
    pin:
      number: ${input_pin}
      mode: INPUT_PULLUP
    internal_filter: 13us # 13us is a limit for ESP32.
    total:
      name: '${friendly_name} - Total gas consumptiom'
      unit_of_measurement: 'm³'
      last_reset_type: auto
      state_class: measurement
      device_class: gas
      accuracy_decimals: 2
      filters:
        - lambda: return x * (1.0 / ${pulse_rate});
```

It ain't pretty, but it works. The transparent "stuff" inside sensor enclosure is hot glue, to keep it in place.

[image]

tiimsvk, 2021-10-08:
> ```yaml
>  pulse_pin: GPIO23
> ```

Hi, could you please send me a simple wiring diagram. Somehow I can't orient myself in that. And English is not my friend. Thanks

Kurisutian, 2021-11-10:
Thanks for the code, which I was using since me reed switch on the gasmeter always was far off. Unfortunately the situation didn't change. I still have about twice as much reading as there is actually gas being used. Do you have an idea what might be causing this and how to get is fixed? Already played with the filtered times but nothing changed. It's also a German gasmeter with their official reed switch probably similar to what you have... I have a fuelcell heating system so this gives me the measured values as well (but only through their cloud system which I prefer not to use if there are other "offline" options) and they are more or less half of the measured values from the reed switch according to their app.

VivantSenior, 2021-12-13:
> Hello In Germany many of the gas meters are non electric but emmit a magnetic impuls e.g 1x per 1/10 m³. I replaced the board with the sensitive LED by a magnetic switch (KY-021 mini) and foud that [...]

Hi,

what about just using Xiaomi Aqara door sensor [MCCGQ14LM](https://zigbee.blakadder.com/Aqara_MCCGQ14LM.html) to detect magnetic impulse?

Btw I've seen two cool use cases for the Xiaomi Aqara door sensors:
- [Chair occupancy sensor](https://community.home-assistant.io/t/diy-zigbee-chair-occupancy-sensor/239517),
- [Rain gauge](https://community.home-assistant.io/t/diy-zigbee-rain-gauge/255379).

Best,
Viv

closed 2022-01-27 by github-actions[bot]

andriuskr, 2022-04-02:
> I just finished my sensor for gas meter so I can share my experience.
> 
> I used [this sensor from local supplier](https://botland.store/magnetic-sensors/14279-reed-switch-module-adjustable-iduino-se015-5903351242189.html), because I grew too impatient waiting for [KY-021 from AliExpress](https://pl.aliexpress.com/item/1005002742977958.html). Both should work the same.
> 
> Positioning for the sensor proved to be somewhat tricky on my meter, but the only real issue was debouncing the reed switch. ESP32 allows up to 13us on `internal_filter` in `pulse_meter` and this is not enough - I got 3-4 times more readings than I should. I found the solution [in this issue](https://github.com/esphome/issues/issues/493#issuecomment-507438088) and it worked great.
> 
> Reed switch D0 pin is connected to pin 23 on ESP32 and configured as `binary_switch`. This switch is used to generate a short pulse on pin 22, which is physically connected to pin 21. `pulse_meter` uses that pin as input, instead of directly relying on input from reed switch.
> 
> So far I got perfect readings.
> 
> Here is my config
> 
> ```yaml
> substitutions:
>   device_name: burn
>   device_description: "Measure your natural gas consumption with the magnetic pulse of your meter."
>   friendly_name: Burn
>   pulse_pin: GPIO23
>   output_pin: GPIO22
>   input_pin: GPIO21
>   pulse_rate: '100' # imp/m³
> 
> esphome:
>   name: '${device_name}'
>   comment: '${device_description}'
>   platform: ESP32
>   board: nodemcu-32s
>   
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> logger:
> 
> ota:
>   password: !secret password
> 
> api:
> 
> text_sensor:
>   - platform: version
>     name: "${friendly_name} - ESPHome Version"
>     hide_timestamp: true
> 
> # Connected directly to input pin used in pulse counter.
> output:
>   - platform: gpio
>     id: gpio_output
>     pin:
>       number: ${output_pin}
> 
> # Used for debouncing reed switch.
> binary_sensor:
>   - platform: gpio
>     name: "${friendly_name} - Pulse input"
>     internal: true
>     pin:
>       number: ${pulse_pin}
>       mode: INPUT_PULLUP
>     filters:
>       - delayed_on: 500ms
>       - delayed_off: 500ms
>     on_press:
>       then:
>         - output.turn_on: gpio_output
>         - delay: 10ms
>         - output.turn_off: gpio_output
> 
> sensor:
>   - platform: pulse_meter
>     name: '${friendly_name} - Gas consumption'
>     pin:
>       number: ${input_pin}
>       mode: INPUT_PULLUP
>     internal_filter: 13us # max for ESP32
>     state_class: measurement
>     device_class: gas
>     unit_of_measurement: 'm³/h'
>     icon: mdi:fire
>     accuracy_decimals: 2
>     filters:
>       - lambda: return x * (1.0 / ${pulse_rate}) * 60;
>     total:
>       name: '${friendly_name} - Total gas'
>       state_class: measurement
>       device_class: gas
>       unit_of_measurement: 'm³'
>       last_reset_type: auto
>       accuracy_decimals: 2
>       filters:
>         - lambda: return x * (1.0 / ${pulse_rate});
> ```
> 
> It ain't pretty, but it works. Sensor enclosure is filled with hot glue to keep things from moving.
> 
> [image]

By any chance, you are still using this ? Cannot get this working, it counts 2-4 x more consumption...

evlo, 2022-07-23:
@landmaj  are you ussing magnetic/metal ring to attach the enclosure? What exactly did you used?

escoand, 2023-02-10:
I can confirm https://github.com/klaasnicolaas/home-assistant-glow/issues/42#issuecomment-919534019 with the dummy binary_sensor, dummy output and the cable between the both is working. Nice would be a built-in solution for this use case.

landmaj, 2025-04-14:
I moved out of that place more than two years ago and don't use the sensor anymore, sorry.

In my case there was an existing product for my meter so I figured out the positioning based on available pictures. Similar to [this one](https://sorbex.pl/sklep/352-nadajnik-impulsow-imp-11.html).