klaasnicolaas/home-assistant-glow #44: KWh meter whit Wemos D1 mini

tiimsvk, 2021-09-06:
is it possible to use [ESP Wemos D1 mini](https://www.aliexpress.com/item/1005002512406896.html?spm=a2g0s.9042311.0.0.27424c4dD22O81)?
If so, can you advise what needs to be adjusted to make it work?
well thank you

klaasnicolaas, 2021-09-06:
Change the `platform` and `board`, same as we advised here: https://github.com/klaasnicolaas/home-assistant-glow/issues/37

tiimsvk, 2021-09-06:
Well thank you.
My question: Where to connect DO on Photodiode?

klaasnicolaas, 2021-09-06:
If I see that schematic, it should indeed be D6.

RubenKelevra, 2021-11-09:
Can we extend the recommended hardware to include the Wemos D1 mini, as it's much more affordable to use them (and we really don't need any ESP32 feature - right?)

m-s-e, 2021-11-09:
> Can we extend the recommended hardware to include the Wemos D1 mini, as it's much more affordable to use them (and we really don't need any ESP32 feature - right?)

i think so - my D1mini has been running for a few weeks now without problems

klaasnicolaas, 2021-11-09:
We've already added the Wemos to the diagram, I think that's enough.

closed 2021-11-09 by klaasnicolaas