klaasnicolaas/home-assistant-glow #46: Not detecting LED of Landis&Gyr ZMB120 meter

Gummientchen, 2021-09-07:
Hi

I can't get the sensor to detect the LED of my meter. I set the sensitivity to the lowest possible without triggering on it's own. On another meter of a neighbor it works without problems.  The ZMB120 uses a green LED instead of a red one, can this be a problem?

I already fixed the wrong polarity of the LED as mentioned in #34 

Has anyone maybe some possible suggestions how I could get this to work? Different sensor or something?

Many thanks.

klaasnicolaas, 2021-09-15:
Doubt if the color of the led has much influence, mine is orange and works fine.

lemppari, 2021-09-22:
Just a wild guess. Since the power led on the photodiode sensor is green and seems to be constantly on, may that could confuse the photodiode somehow. Could you try to tape the power led so it won't be visible and leak to the photodiode?

yvansandoz, 2021-10-09:
Hello my meter is a Kamstrup as shown on the picture, but same problem, I cannot detect the LED of the meter. I did test my system with another light source and it works. Any adjustments (software or hardware) that can be done?

huskyte, 2022-07-09:
Just came across your issue as I am going to install one on my Landis Ger. Is the LED really green? I just found one manual stating it would be IR. I will come back to you as soon as I get mine up and running (or not).

huskyte, 2022-07-09:
Manual: "The test diode is used for testing the meter. It transmits infrared pulses cor-responding to the current measured value. The number of pulses per unittime depends on the meter constant set (according to the inscription on thedial, see also Fig. 3.2) and on the power applied." This is a ZxD120AP. I hope the LED is sub 1000 nm as a photodiode usually has 1000+ nm sensitivity.
Furthermore:
"The infrared test diode on the meter below the LCD should be used formeter testing. This supplies pulses at a frequency dependent on the meterconstant R and the power present, whereby the rising edge is always deci-sive for the test.Note that the digital signal processing provides a delay of 2 seconds be-tween the instantaneous power at the meter and the appearance of thepulses at the test diode. No pulses are lost.The number of pulses per second for the desired power is obtained bymultiplying the meter constant R by the power in kW divided by 3600.Example:    Meter constant R = 1000Power P = 35 kWf-LED = R x P / 3600 = 1000 x 35 / 3600 = 10 imp/sThe test diode is continuously lit at creep."
Enough theory, lets see next wekk what it does...

huskyte, 2022-07-16:
Ok, checked it. Mine is working perfectly fine. No sensitivity adjustment needed, all working and reading precisely without glitch. Porting it now to a M5Stick with dual sensor to read the aux ZMB of our house.

realfire23, 2022-09-02:
same problem here with Landis&Gyr  E220 and EHM
with 10000 imp

Used 3 photodiodes. wemos,nodemcu etc  Other light sources work good for detection so the photodiodes seem to work fine.

On the Landis& Gyr e.g. the diode blinks permanently so it receives impulses.

closed 2023-01-05 by klaasnicolaas