klaasnicolaas/home-assistant-glow #47: I get very strange constant  Sending state 450000.00000 W  readings.

geeneo, 2021-09-08:
Hi

I get very strange constant  Sending state 450000.00000 W  readings, anyone also get this. Not sure where to begin.

See my debug log,

[D][sensor:131]: 'House - Power consumption': Sending state 2310.65454 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.05300 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2298.85059 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.05400 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.05400 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2282.81543 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.05600 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2282.81543 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.05800 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy
**[D][sensor:131]: 'House - Power consumption': Sending state 2271.29346 W with 0 decimals of accuracy**
[D][sensor:131]: 'House - Total energy': Sending state 0.05900 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06100 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06300 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2271.29346 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06400 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06400 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2282.81543 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06600 kWh with 3 decimals of accuracy
**[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy**
[D][sensor:131]: 'House - Power consumption': Sending state 2271.29346 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06700 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.06800 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2282.81543 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.07000 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2282.81543 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.07200 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 450000.00000 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Power consumption': Sending state 2271.29346 W with 0 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.07300 kWh with 3 decimals of accuracy
[D][sensor:131]: 'House - Total energy': Sending state 0.07300 kWh with 3 decimals of accuracy

KixMan28, 2021-09-08:
Please check issue #34.

dominicknoetze, 2021-09-08:
Try setting the sensitivity on you photo diode.I had a similar problem and fine tuning it solved my problem.

mrwhite31, 2021-09-11:
I am also getting this high value readings regularly. 600.000W, 450.000W, 900.000W... Every flash on the meter is detected correctly and the red led is flashing. Not sure why I am getting those glitches. How did you do the fine tuning on the photodiode?

KixMan28, 2021-09-11:
@mrwhite31 you have a potentiometer on your module (blue box with gray screw top). Try adjusting that. Please check your photodiode if it soldered correctly. Many of us had the photodiode soldered in reverse. Check issue #34 for details.

mrwhite31, 2021-09-11:
@KixMan28 thanks. I've alteady adjusted the potentiometer to the lowest possible sensitivity that it is still catching the flashes. Isn't that correct? 
the diode is soldered the correct way. Also tried other modules.
I forgot to mention that it started not too long ago. It ran fine and without glitches for one and a half month before

KixMan28, 2021-09-11:
@mrwhite31 Sorry, I cannot help you further. I had similar issues when my photodiode was soldered in reverse. Since I re-soldered it everything works as intended.

jLynx, 2021-09-12:
Try adding this 
```internal_filter: 100ms```

mrwhite31, 2021-09-13:
@jLynx adding internal_filter solved it, thanks!

klaasnicolaas, 2021-09-15:
I think we can close the issue then?

jLynx, 2021-09-15:
@klaasnicolaas please close this issue

closed 2021-09-15 by klaasnicolaas