klaasnicolaas/home-assistant-glow #48: [REQUEST] Water flow metering of Kamstrup smart water meters via IR optical eye?

Hedda, 2021-09-08:
Would love water flow reading support for smart water flow meter added to Home Assistant Glow hardware if it could be possible?

Not quite sure if all newer water meters offer an LED which pulses base on a set consumption value or not? See probe examples:

https://www.mysensors.org/build/pulse_water

https://www.kompf.de/tech/watir.html

However believe some modern water meters offer a two-way bidirectional infrared port (IR receiver and sender) for communication?

https://www.mysensors.org/build/ir

http://jheyman.github.io/blog/pages/WirelessWaterMeter/

As I understand those optical probes should work with optical infrared waves be compliant with the  international standard IEC 62056-21 (formerly IEC 1107) or ANSI C12.18 communications protocols for energy metering for reading utility meters?

https://en.wikipedia.org/wiki/IEC_62056

https://en.wikipedia.org/wiki/Smart_meter#Protocols

A hackerspace group in Denmark is selling these IR reading heads for about $14 US (Google translate as test written in Danish) and they wrote a note that using magnets is required as the IR port transmitter / reader eye does not activate unless it sense a magnet:

https://wiki.hal9k.dk/projects/kamstrup

I believe that Kamstrup water meters with IR optical head are very common in Europe:

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters/multical-21

[image]

[image]

I myself live in Sweden where I think Kamstrup MULTICA 21 / Kamstrup flowIQ 210x ultrasonic water meter is the most common model in normal private residential houses here in Scandinavia (Sweden, Norway, Denmark, Finland, and Island).

Kamstrup official (and expensive) optical read-out head tools are called 6699-099 (USB) and 699-102 (9F D-sub plug serial)

https://www.termonet.rs/pdf/Kamstrup/Optical%20Read-out%20(IR)%20Head%20-%20Data%20Sheet%20-%20English.pdf

Kamstrup apparently call their implementation for Kamstrup Meter Protocol (KMP) and someone has reverse-engineered it here:

https://github.com/bsdphk/PyKamstrup

https://frack.nl/wiki/Stadsverwarming

https://forum.mysensors.org/topic/3525/district-heating-city-heating-stadsverwarming-mysensor-ir-sender-receiver/

https://www.domoticz.com/forum/viewtopic.php?t=11333

They currently also have Kamstrup flowIQ 2200 "advanced model with acoustic leak detection": 

[image]

[image]

and Kamstrup flowIQ 3100 "Commercial and industrial" model 

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters/flowiq-2200

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters/flowiq-3100

Kamstrup seem to provide technical documentation about all their products here:

https://products.kamstrup.com/index.php

PS: Off-topic but FYI; some if not all of these models of Kamstrup water meters can also be read remotely via Sigfox network RF (Radio Frequency) but that communication traffic is most of the time encrypted, at least if the water meter itself is owned and operated by a utility company or a service provider.

https://www.kamstrup.com/en-en/water-solutions/water-meter-reading

https://www.kamstrup.com/en-en/water-solutions/water-meter-reading/usb-meter-reader

If the case is that the RF radio communication is not encrypted then there look to be projects that can work with that:

https://github.com/adams-okode/kamstrup-integration

https://github.com/weetmuts/wmbusmeters

https://github.com/tobiasrask/wmbus-client

klaasnicolaas, 2021-09-08:
IR interface is quite different from what the Glow uses. In addition, I think that this is more something for a completely new project and this does not belong to the Glow project.

Hedda, 2021-09-08:
> IR interface is quite different from what the Glow uses. In addition, I think that this is more something for a completely new project and this does not belong to the Glow project.

OK, suspected that might be the case, closing as I created a new request to ESPHome here -> https://github.com/esphome/feature-requests/issues/1402

closed 2021-09-08 by Hedda

Hedda, 2021-09-08:
FYI, most of the older and newer Kamstrup electricity meters look to feature a similar or same type of bi-directional optical header.

That includes the one that you yourself look to use, see:

[image]

klaasnicolaas, 2021-09-08:
Of course I can look at it in the future, but for now I'll stick to the pulse led.

Hedda, 2021-09-08:
FYI, there is an interesting active discussion about SML (Smart Message Language) protocol here -> https://github.com/esphome/feature-requests/issues/1041