klaasnicolaas/home-assistant-glow #54: Last reset type

markinsutton, 2021-09-16:
last_reset_type has been removed since 2021.9.0 so I am getting the below message when I change to this data is no longer being recorded in Home assistant energy dashboard. 

state_class: total_increasing should be used for total values.
    last_reset_type: auto

made the following changes and no longer receiving data into Home Assistant 

https://github.com/markinsutton/home-assistant-glow/blob/patch-1/home_assistant_glow.yaml

klaasnicolaas, 2021-09-16:
See the latest release 😉 

https://github.com/klaasnicolaas/home-assistant-glow/releases/tag/v2.0.1

closed 2021-09-16 by klaasnicolaas

markinsutton, 2021-09-17:
I know you closed this issue but since updated my yaml to v2.0.1 the values are showing in home assistant but not showing in the energy dashboard any idea why? I guess it working for everyone else. ?

klaasnicolaas, 2021-09-17:
that shouldn't be something related with the changes in the last release, it works fine here.

KixMan28, 2021-09-17:
It works fine fir mee too.

markinsutton, 2021-09-17:
Okay very strange as all the entities show in Home Assistant. was working day before just updated the 2 lines in Yaml file and by  the morning I had no sensor data in energy dashboard. but all the ESP home data shows energy usage. will have to do some research.