klaasnicolaas/home-assistant-glow #55: zero consumption on  electricity meter red Led always on

tiimsvk, 2021-09-16:
When the circuit breaker on the electricity meter is switched off, the red LED is always on. Is this solved in yaml code?

klaasnicolaas, 2021-09-16:
Do you run the latest release? V2.0.1

tiimsvk, 2021-09-17:
Yes I use. My question:
it is incorporated in the code.
Or is it possible to work?

I don't mean completely in the code so I ask?

klaasnicolaas, 2021-09-17:
Wait you mean the pulse led on your meter or the debug led on the glow?

tiimsvk, 2021-09-17:
I mean an electricity meter.  When the main circuit breaker is turned off, the red LED of the electricity meter starts to light up.

klaasnicolaas, 2021-09-20:
Which electricity meter do you have? I've never experienced this before that a led stays on.

tiimsvk, 2021-09-21:
This mi problem
[https://www.youtube.com/watch?v=S-bftSD92bg](url)
zero consuption energy led red always on

toberhof, 2021-11-09:
Mine does the same. It is expected behavior if there is no energy consumption the led will be constant red. It also appears when my solar panels produce more energy than the home can use

klaasnicolaas, 2021-11-09:
The only question is whether and how we can make something for this, I don't really know a solution at the moment 🤔

4ordy, 2021-11-19:
I don't think this is a bug, I too like @toberhof was checking out my meter and found this thread researching whether Home Assistant Glow would still work event if I was exporting to the grid from my solar (led on) 
Please don't change it but call it out in the docs alternatively include a setting whereby you set something if have solar which exports and the led remains on.

blomnik, 2021-12-27:
Yes, I have a similar meter. Idle state of meter's onboard LED is on. This mean, that it shortly turns off when a "pulse" occurred.  Os it possible to use current release without any modifications?

klaasnicolaas, 2021-12-27:
Does this need to be addressed further? As in what is the behavior currently in the measurements and what would we like it to eventually do and is this something related to the Glow or more in the integration we use from ESPHome?

Don't think I can help a lot unfortunately, in my country it is not common that there is still a switch in front of the meter and that makes testing a bit more difficult.

blomnik, 2021-12-27:
It will be convenient if ot will be possible to switch from rising to falling edge detection. I think, this will solve a problem.

klaasnicolaas, 2021-12-27:
Those kinds of changes should then be addressed in the esphome repository for the pulse_meter component 🤔

Borcon, 2021-12-28:
@klaasnicolaas 
i have the same issue as @blomnik .
My electricity meter (3 years old) is from EON and the pulse light is default on. It turns shortly off when a pulse is occurred.
In Germany many people have this newer electricity meter.
If I understand it correctly, then it is currently not possible with this meter? Or i am wrong?

blomnik, 2021-12-29:
@Borcon, actually, I use this without any changes in sources with small modification in YAML file. It works, but I dont know yet is it works correctly. Still in debug.

klaasnicolaas, 2021-12-29:
@blomnik What adjustments have you made?

Borcon, 2022-01-02:
@blomnik In my case i have no detection. 😔 The value remains at zero.

closed 2022-02-09 by github-actions[bot]