klaasnicolaas/home-assistant-glow #58: Is it normal for the bright green LEDs on the photodiode to be on?

tejasitraj, 2021-09-19:
Is it normal for the pwr-led on the photodiode to be always on? I feel like my energy meter's LED has no chance against that bright green LED...

And when i increase the sensitivity of the diode, the DO-LED comes on as well..
[image]
.

moorep999, 2021-09-20:
Mine does the same. Power always on, and the other when the photodiode detects light. The 3d printed enclosure for the diode has a partition between the LEDs and the diode to help stop stray light. Compared to my meter’s LED, they’re much brighter.

Can you use some duct tape/cardboard to isolate the diode and give it a better chance of detecting?

jurgenweber, 2021-09-27:
was going to ask the same question, since the LED flash from my meter is so soft I was concerned about this.. but I also think both my lights are on all the time. Sounds like this is incorrect.

RubenKelevra, 2021-11-16:
> Mine does the same. Power always on, and the other when the photodiode detects light. The 3d printed enclosure for the diode has a partition between the LEDs and the diode to help stop stray light. [...]

So your module looks the same and is working fine? I have trouble getting mine to recognize the light from my meter, while I cranked the sensitivity up to the maximum where the DO-LED stays off when it's in total darkness.

RubenKelevra, 2021-11-16:
> was going to ask the same question, since the LED flash from my meter is so soft I was concerned about this.. but I also think both my lights are on all the time. Sounds like this is incorrect.

I think DO-LED means that the module sends a digital 1 out, so "light is there". So it should be off as long as there's no flash from the meter - as far as I understand.

moorep999, 2021-11-16:
> So your module looks the same and is working fine? I have trouble getting mine to recognize the light from my meter, while I cranked the sensitivity up to the maximum where the DO-LED stays off when [...]

Turns out my photodiode was installed backwards. Came across a random comment a while ago. They’ve now referenced it in the main glow page (issue 34)
https://github.com/klaasnicolaas/home-assistant-glow

RubenKelevra, 2021-11-16:
Hm I've checked that. Let me double check that :)

klaasnicolaas, 2021-11-18:
I think the main question has already been answered. Yes, this LED is always on and the case tries to block the light as much as possible so that it does not affect the sensor. Can we close the issue?

tejasitraj, 2021-11-18:
Absolutely, thank you!

closed 2021-11-18 by tejasitraj