klaasnicolaas/home-assistant-glow #60: Strange values from the pulse_meter

adorobis, 2021-09-24:
I can't figure out what I'm doing wrong. The pulse_meter does not count pulses (W) and the total energy increase for each pulse is much higher than expected. Here is the log:
```
[21:49:13][D][sensor:121]: 'Meter - Total energy': Sending state 0.13467 kWh with 3 decimals of accuracy
[21:49:17][D][binary_sensor:036]: 'Meter pulse': Sending state OFF
[21:49:17][D][sensor:121]: 'Meter - Total energy': Sending state 0.15400 kWh with 3 decimals of accuracy
[21:49:17][D][binary_sensor:036]: 'Meter pulse': Sending state ON
[21:49:17][D][sensor:121]: 'Meter - Total energy': Sending state 0.16333 kWh with 3 decimals of accuracy
[21:49:20][D][binary_sensor:036]: 'Meter pulse': Sending state OFF
[21:49:20][D][sensor:121]: 'Meter - Total energy': Sending state 0.19000 kWh with 3 decimals of accuracy
[21:49:20][D][binary_sensor:036]: 'Meter pulse': Sending state ON
[21:49:24][D][binary_sensor:036]: 'Meter pulse': Sending state OFF
[21:49:24][D][sensor:121]: 'Meter - Total energy': Sending state 0.21467 kWh with 3 decimals of accuracy
[21:49:24][D][binary_sensor:036]: 'Meter pulse': Sending state ON
[21:49:26][D][pulse_counter:159]: 'Pulse Counter': Retrieved counter: 0.00 pulses/min
[21:49:26][D][sensor:121]: 'Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[21:49:28][D][sensor:121]: 'Meter - Total energy': Sending state 0.24067 kWh with 3 decimals of accuracy
[21:49:32][D][sensor:121]: 'Meter - Total energy': Sending state 0.26600 kWh with 3 decimals of accuracy
[21:49:35][D][sensor:121]: 'Meter - Total energy': Sending state 0.29333 kWh with 3 decimals of accuracy
[21:49:39][D][binary_sensor:036]: 'Meter pulse': Sending state OFF
[21:49:39][D][sensor:121]: 'Meter - Total energy': Sending state 0.32267 kWh with 3 decimals of accuracy
[21:49:39][D][binary_sensor:036]: 'Meter pulse': Sending state ON
[21:49:43][D][sensor:121]: 'Meter - Total energy': Sending state 0.35267 kWh with 3 decimals of accuracy
[21:49:47][D][sensor:121]: 'Meter - Total energy': Sending state 0.38200 kWh with 3 decimals of accuracy
[21:49:50][D][sensor:121]: 'Meter - Total energy': Sending state 0.40667 kWh with 3 decimals of accuracy
[21:49:54][D][binary_sensor:036]: 'Meter pulse': Sending state OFF
[21:49:54][D][sensor:121]: 'Meter - Total energy': Sending state 0.43267 kWh with 3 decimals of accuracy
[21:49:54][D][binary_sensor:036]: 'Meter pulse': Sending state ON
[21:49:58][D][binary_sensor:036]: 'Meter pulse': Sending state OFF
[21:49:58][D][sensor:121]: 'Meter - Total energy': Sending state 0.45800 kWh with 3 decimals of accuracy
[21:49:58][D][binary_sensor:036]: 'Meter pulse': Sending state ON
[21:50:02][D][sensor:121]: 'Meter - Total energy': Sending state 0.48333 kWh with 3 decimals of accuracy
[21:50:02][D][sensor:121]: 'Meter WiFi Signal': Sending state -68.00000 dBm with 0 decimals of accuracy
[21:50:05][D][sensor:121]: 'Meter - Total energy': Sending state 0.50867 kWh with 3 decimals of accuracy
```
I've added a binary sensor to the gpio pin to monitor its behavior in HA.
My pulses/kWh is 1500.
Strange thing is that when meter diode is off the binary sensor is On and when the diode lights up the binary sensor is turned off (maybe it should be this way?).
Here is my config:
```
substitutions:
  device_name: meter
  friendly_name: Meter
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO4
  # imp/kWh rate #
  pulse_rate: '1500'

esphome:
  name: '${device_name}'
  platform: ESP8266
  board: nodemcuv2

binary_sensor:
  - platform: gpio
    pin: ${pulse_pin}
    name: "Meter pulse"
    id: meter_pulse
    
sensor:
  - platform: wifi_signal
    name: "Meter WiFi Signal"
    update_interval: 60s
    
  - platform: pulse_meter
    name: '${friendly_name} - Power consumption'
    id: pulse_meter_id
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * (60.0 / ${pulse_rate} * 1000);
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});
```
Am I doing something wrong?

adorobis, 2021-09-24:
OK, I've just realized that the module I have has an LDR (that's what they've had locally, not waiting several weeks for delivery from China). So the question is if it can work with the LDR?

klaasnicolaas, 2021-09-24:
I don't see anything weird in your logs.

I think your expectations are more in line with the `pulse_counter` component of ESPHome, but the `pulse_meter` is a combination of the `pulse_counter` and `pulse_width` component. So it not only looks at the fixed amount of energy that it should add up, but also determined the value based on the time between the pulses. That is why each measurement in terms of power (W) and counting the total energy (kWh) per pulse can be different.

klaasnicolaas, 2021-09-24:
What is you use this: (1000.0 instead of 1000)
`- lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);`

adorobis, 2021-09-24:
> What is you use this: (1000.0 instead of 1000)
> `- lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);`

Changed it and there is no difference. The Power sensor is not getting updated and total energy updates are too high.
I've created some template sensors in HA based on just the `binary_sensor` from esphome and they are much more accurate (I just add 1/1500 kWh with every meter pulse to the total energy sensor).
I have also created a `pulse_counter` sensor in esphome and strangely it is not showing any value neither - stays all the tiime on 0.000. Could this be that it is because the normal level of the gpio is high and goes low when the meter led blinks?

klaasnicolaas, 2021-09-24:
Maybe you need to add this line: https://github.com/klaasnicolaas/home-assistant-glow/blob/master/home_assistant_glow.yaml#L100

adorobis, 2021-09-24:
> Maybe you need to add this line: https://github.com/klaasnicolaas/home-assistant-glow/blob/master/home_assistant_glow.yaml#L100

Looks like that was it! Many thanks. I'll monitor it for a while and report back. If it works it also means that LDR is also good for this project.

klaasnicolaas, 2021-09-24:
Would still not recommend to use the LDR, because of the reliability in terms of measurements. But in the end it is certainly possible to use an LDR.

adorobis, 2021-09-24:
Anyway I've ordered the photodiode but only available in China so it will take a while. Not sure what is the problem with the photodiode modules availability...

closed 2023-08-14 by adorobis