klaasnicolaas/home-assistant-glow #62: No pulse detected

olskar, 2021-09-27:
Hello!

I get not pulse detected even though I have put the diode in front of the red blinking led on the power meter. Perhaps I should try change the sensitivity? In what direction do I turn the potentiometer to increase the sensitivity?

enshoe, 2021-10-03:
Hey,
I have the same issue. I have tried turning the potentiometer at intervals both up and down. No dice.
Shorting the legs on the diode produces a red flash on the LED. Perhaps a faulty diode or potentiometer for us both?

RubenKelevra, 2021-11-16:
I'm having the same issue. My diode module has a digital out led (which is green) and a power-led which is also green. I taped them off to avoid any feedback from them.

But I can't get it to work... I turned the sensitivity down exactly to the point where the digital out led turns off in total darkness and no luck detecting the flashing LED from the meter (which is actually pretty bright).

So I'm wondering if there's a limit in color range what those LEDs can detect. I mean, maybe it's just not sensitive to a red light?

klaasnicolaas, 2021-11-16:
@RubenKelevra Have you checked if this issue applies to you? https://github.com/klaasnicolaas/home-assistant-glow#wrong-soldered-diode. As far as I know, a red and orange LED should work just fine.

RubenKelevra, 2021-11-16:
Yeah the diode is plugged into the board correctly and I've tested it with a "white" flashlight, for bright light it works fine.

RubenKelevra, 2021-11-17:
Okay, the diode is correctly mounted - but the module seems to be broken. Anybody got any idea if something is worth testing, or just throw the module in the "art" bucket and order a new one?

tausen, 2021-11-17:
It would be interesting to hear whether it is defective or just responds poorly to your diode -- perhaps we have the same issue. I've ordered an extra module and will compare the two with my meter.

Do you have some instrument available like an [ADALM2000](https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/adalm2000.html) or [Analog Discovery](https://digilent.com/shop/analog-discovery-2-100ms-s-usb-oscilloscope-logic-analyzer-and-variable-power-supply/)? An ADC pin on the ESP32 could probably do the job as well, perhaps [this project](https://github.com/BojanJurca/Esp32_oscilloscope)? I could see a clear drop in voltage on the A0 pin, but could not adjust the sensitivity to a good spot.

RubenKelevra, 2021-11-18:
> It would be interesting to hear whether it is defective or just responds poorly to your diode -- perhaps we have the same issue.

Well, I can trigger a 1/0 event with the sensitivity poti, but the diode seems to do nothing. Even from a dark room to a bright light directly shining in it there's no change.

> Do you have some instrument available like an [ADALM2000](https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/adalm2000.html) or [Analog [...]

Only an el-cheapo multimeter... but it can measure diodes ... if this is of any help?

tausen, 2021-11-19:
> Even from a dark room to a bright light directly shining in it there's no change.

Oh, okay -- in that case I guess something is broken... 

It might not be worth the effort, but if you want to dig you could start by measuring the voltage on the A0 output with your multimeter. It should be about 3.3V when in darkness and drop to 0V when exposed to bright light. This comes directly from the diode, so if there's nothing there try the diode pins instead. 

If the diode and A0 seem OK, maybe check connections on the opamp. VCC should be 3.3V, IN+ should be the same as A0 (the diode output), IN- should depend on the potentiometer and OUT should be the same as D0:

[image]

sfritz67, 2021-12-25:
Hello,
I also have the same issue.
Photodiode works with my phone light, but do not react with power meter orange light.
I tried different potentiometer positions, without any succes.

@tausen : You told us about a comparator on A0, how do you use it?

Is there another Photodiode  that could work with an orange light?

tausen, 2021-12-26:
Here's what I did, @sfritz67:

      5V o───────────────────────────────┐
                TL072CP                  │
             ┌────────────┐              │
    ┌──────┬─│ 1OUT  Vcc+ │──────────────┤
    │      └─│ IN-   2OUT │───o OUT     ┌┴┐
    │ IN o───│ IN+   2IN- │──────────┐  │ │R1
    │GND o─┬─│ Vcc-  2IN+ │─┐        │  └┬┘
    │      │ └────────────┘ │        └───┤
    └───── │ ───────────────┘           ┌┴┐
           │                            │ │R2
           │                            └┬┘
           └─────────────────────────────┘                     

(sorry for the crude drawing)

So this is just a buffer amplifier on the input (which shouldn't be needed anyway) followed by a basic comparator with threshold set by R2/(R1+R2). After some testing I landed at R2 = 6.8k and R1 = 4.7k, which yields a threshold about 2.96 V.

IN is connected to the A0 pin on the diode breakout board, and OUT goes to the ESP32. I took GND and 5V directly from the ESP32.

I used the TL072CP opamp because that's what I had lying around - if you have something else, odds are that'll be fine too. Just check the pinout in the datasheet and make sure that the high-level output is somewhere near but does not exceed 3.3V. If you have a rail-to-rail opamp you can use 3.3V on `Vcc+` instead of 5V. If you have a single-comparator IC, just skip the buffer opamp.

Does that make sense?

> Is there another Photodiode that could work with an orange light?

I don't know. It would be interesting to try out some other diodes, but I'll have to go ebay shopping first...

raforo, 2022-01-06:
I've bought three photodiode at Banggood, they all arrived with the polarity changed, I soldered correctly, but I cannot adjust the sensitivity to the red meter's LED, it seems that it is very dim and goes from giving a continuous signal in DO to give no signal when moving the variable resistor.
Everything else works perfectly, but I cannot adjust the sensitivity of the Led , a strong light like a flashlight it's detected, but the LED does not.

raforo, 2022-01-12:
> I've bought three photodiode at Banggood, they all arrived with the polarity changed, I soldered correctly, but I cannot adjust the sensitivity to the red meter's LED, it seems that it is very dim [...]

I've replaced the photodiode by an LDR on the Bangood board and it has started working properly, now it detects the pulse of the red LED of the meter without problem.

Thanks

sfritz67, 2022-01-14:
> I've replaced the photodiode by an LDR on the Bangood board and it has started working properly, now it detects the pulse of the red LED of the meter without problem.
> 
> Thanks

Thank you for your feedback, I tried too, but it did not work for me. Maybe I need to adjust the potentiometer a bit more.
I don't have the specs for the one I used, it's an old one I bought for another project few years ago... I need to check the resistance range to see if it can be the issue.

What kind of LDR did you use? do you know the specification for the one you used?

raforo, 2022-01-14:
> > I've replaced the photodiode by an LDR on the Bangood board and it has started working properly, now it detects the pulse of the red LED of the meter without problem.
> > Thanks
> 
> Thank you for your feedback, I tried too, but it did not work for me. Maybe I need to adjust the potentiometer a bit more. I don't have the specs for the LDR I used, it's an old one I bought for another project a few years ago... I need to check the resistance range to see if it can be the issue.
> 
> What kind of LDR did you use? do you know the specification for the one you used?

I've bought it in a local spanish store, but it's similar  to this one
https://es.aliexpress.com/item/1005002568306221.html
I hope this help you

iamvlanone, 2022-01-25:
I have the same issue with no pulse detection or unreliable detection. I found that if the lights are on in the room it detects the pulses of the dim red LED of my utility meter. I tried to set the sensitivity and could not get it to work reliably. I accidentally "bridged" the LED pins with my finger and found it would make it detect properly. I started looking at some sort of resistance/current limiting issue and the 10k pull up resistor on pin 5 of the LM393 caught my attention. I replaced the resistor with a 12k and then a 15k resistor and both made it more sensitive. My suspicion is that the diode cannot drop enough voltage to pull pin 5 to ground and therefore reducing the current flowing through the pull-up resistor will make it drop enough voltage. Any ideas/explanations will be welcome.

I am busy testing it in a dark room and will check back after a few hours to see if it was detecting reliably.

I am using the Banggood photodiode module that came with a reverse soldered diode and fixed it. below is the resistor that I changed. I will update my findings. I hope this can help someone dealing with the same issue.
[image]

crazytok, 2022-02-15:
> I have the same issue with no pulse detection or unreliable detection. I found that if the lights are on in the room it detects the pulses of the dim red LED of my utility meter. I tried to set the sensitivity and could not get it to work reliably. I accidentally "bridged" the LED pins with my finger and found it would make it detect properly. I started looking at some sort of resistance/current limiting issue and the 10k pull up resistor on pin 5 of the LM393 caught my attention. I replaced the resistor with a 12k and then a 15k resistor and both made it more sensitive. My suspicion is that the diode cannot drop enough voltage to pull pin 5 to ground and therefore reducing the current flowing through the pull-up resistor will make it drop enough voltage. Any ideas/explanations will be welcome.
> 
> I am busy testing it in a dark room and will check back after a few hours to see if it was detecting reliably.
> 
> I am using the Banggood photodiode module that came with a reverse soldered diode and fixed it. below is the resistor that I changed. I will update my findings. I hope this can help someone dealing with the same issue. [image]
> 
> Update: So far I have been getting pretty accurate detection with various loads.

Im having the same problems with my sensor trying to read the led on a Kamstrup 382MxC. It doesn't matter how Im trying to adjust the potentiometer, It wont read the led properly. So I'm really curious if you still find this working propertly? If so, what resistor do you suggest trying, 12k or 15k ?

May I ask what power meter you are trying to read with home assistant glow?

fokodk, 2022-09-11:
I got it working by replacing the LED with a photoresistor, unfortunately I don't know its specs as it came with my elegoo r3 uno kit 
[image]

ronjtaylor, 2022-12-07:
My glow pulse stopped working after working for a few weeks.
We had had a number of very hot days and I think my Double Sided tap might have slipped.

I had used the 3D print files and had a plastic case made for the Photodiode board. 
[image]
I did not realise until i did some research that the light entering a photodiode should come from the side
[image]
With the design of this case, the diode's top is exposed to the pulse and the sides would only get reflected light in the hole.
To make mine work again I had to be very particular of the mounting of the 3D Case with double sided tape and use Duct Tape to restrict any ambient light from the enclosure to enhance the pulse light from the meter as well as giving the double sided tape a helping hand in staying in position in heat.
Hopefully it will continue to work! 
I am using GLOW to validate a Shelly EM that I read consumption on Ch 1 and Solar Generation on Ch 2
NET = Consumption - Generation
Anyway I just thought I would pass that along.

Onaka, 2023-01-13:
I think you'll find that if you're having trouble detecting the pulses with your photodiode board, you'll want to swap out the 10k resistor between VCC and diode+, with a 22k one, mine started working like a charm after I made that modification.

I bought the H Hilabee ones off of Amazon, you'll identify them based on the fact that there are 3 resistors flanked by a capacitor on each side right behind the diode. The board underside also says "QIFEI".
In my unit, swapping out R2 for a 22k resistor fixed all issues.

I'm pretty sure the circuit is laid out the same on all of these variations, but they're maybe using different diodes or different values for the resistors, in any case, if it's got an LM393, I'm pretty sure this modification should work.

Obviously take with a grain of salt, I'm not responsible for your house burning down or loss of sensor or anything of the sort... I'm a random dude on the internet, don't do what I'm suggesting unless you have some inkling about what you're doing.

AndyPook, 2023-03-16:
> I found that the photo diode is most sensitive to light falling on the end (through the lens) and not very sensitive to light falling on the sided. It's an awkward mount but placing the photo diode end-on to the front of the meter works well for me. The red light from the LED in the meter passes directly into the top end of the photo diode.
> 
> [image]

This was the fix for me (thanks).
The tip of the photodiode seems to act as a lens.

The cabinet my meter is in would not give enough space for this.

My solution was to bend the legs of the photodiode to angle it towards the meter.
I wanted to be able to see the led's on the board, so I bent it towards the "bottom" of the board. The legs aren't very long, so it may not be able to bend it over a full 90 degrees. But even just 45 is enough to point it at the led on the meter.

Or just bend the photodiode "upwards", away from the board and mount the board "facing" the meter <shrug>

I guess this must be an assumption when using the custom box defined in the repo?

ronjtaylor, 2024-01-25:
Close it from my point of view

olskar, 2024-01-27:
> @olskar Is the original issue still relevant, or can I close it?

I am not using anymore, so no idea, I guess can be closed :)

closed 2024-01-27 by olskar