klaasnicolaas/home-assistant-glow #77: HA Glow and HA Utility Meter issue

nefelodamon, 2021-11-01:
Hi All,

Having an issue with the utility meter sensors in HomeAssistant.
When the consumption in my house is normal, everything works OK. 

Once in a while though we put our electric car to charge and the consumption rises to 7KW per hour for several hours. After a while though the utility meters will stop recording the information from HA Glow.

Here is an example of a hourly utility meter. You can see that:

- from 23:00 to 24:00 is OK
- from 24:00 to 24:51 is fine, but then stops and flatlines. I need to restart HA to get it to start logging OK again.

[image]

I am suspecting that the HA Glow, just sends too many values and the recorder cannot keep up.
I have 5 utility meters that all get the same info from HA Glow sensor for long term statistics (hourly, daily, weekly and yearly).

Any ideas what might be the issue here?

klaasnicolaas, 2021-11-01:
Based on the data, I can't conclude whether this is an issue for the Glow or whether you're better off in the core repository.

nefelodamon, 2021-11-02:
Hi Nicolas, 

Thanks for your reply.

I was thinking perhaps if there is a way to send the value every x seconds perhaps so I can see if that helps?
It seems 7500 events per hour (125 per minute) are too much for HA recorder to handle properly.

I am going to raise that in HA forum as well and see if anyone can help there.

klaasnicolaas, 2021-11-02:
Are the 125 events per minute only from the glow?

nefelodamon, 2021-11-02:
> Are the 125 events per minute only from the glow?

Correct.... My car is charging at 7.5Kw per hour and my meter rate is 1000imp per KWh. So it blinks about 2 times per second or at least 125 times per minute.

In the beginning I can see that the Glow Total Energy sensor in HA is increasing by 1 very fast (as Glow sends an updated value every time it sees a blink). That is all fine though for a while but HA stops recording events some time after (note that all sensors are affected).

I suppose that this high rate may not be that big issue for most people, but I have created 5 utility meters in HA so the events actually may be multiplied by 5 (since all utility meters track the Glow Total Energy sensor).
 
At the moment after adding the -heartbeat: 30s to the filter, the total energy value is send every 30 seconds. Glow device keeps track of the blinks in these 3 seconds and sends the total in that specified interval.

I need to drive the car for a while now before I charge it again, but I am pretty sure it will resolve the issue as this cuts down massively the amount of events generated in HA.

If this works for me, perhaps a commented out line can be added for this heartbeat option with a corresponding entry in the documentation as others may need to limit the event rate as well.

nefelodamon, 2021-11-20:
So this seems to fix the issue with recorder.
If anyone needs to limit the rate that events are send to HA you can add the - heartbeat: 30s option.

closed 2021-11-20 by nefelodamon