klaasnicolaas/home-assistant-glow #97: Total energy usage for today only

Beenay25, 2021-12-14:
I may be missing something obvious, but is it possible to get the energy used today only?

I can get the current energy usage, and the total energy usage. But the total seems to be for all time (I'm currently showing 8kwh total where my actual usave is about 2.5kwh for today.

Is there a way to get it to reset this total value at midnight?

Beenay25, 2021-12-14:
Hi,

Thanks for adding the new-feature flag.

If you're adding this as something new, it might be handy to be able to state when the count resets. Like midnight, or 4am etc...

klaasnicolaas, 2021-12-14:
I never added the day entity because the total kWh entity for the energy dashboard is enough, but can still add it eventually. The reset happens by default at midnight, I don't know if we can change that much 🤔

Beenay25, 2021-12-14:
Is there a way for us to manually reset the value? An API call or something?

It's an excellent product. So easy to get running :)

klaasnicolaas, 2021-12-14:
Maybe we can make an extra reset service for the day entity 😄

klaasnicolaas, 2021-12-15:
I'm trying some things out with the new options in 2021.12 
[image]

closed 2022-01-01 by klaasnicolaas