klaasnicolaas/home-assistant-glow #100: Block my pi  during compile

caaveiro, 2021-12-19:
Hello.

I have completely done this project (3d printing / printed circuit) but I run into a problem when trying to install it from EspHome.

With the file home_assistant_glow.yaml it blocks my opi when trying to compile it. Can there be any problems with the configuration or dependencies of the file?
Any other project I do in esphome always works for me and compiles perfectly.

The log is as follows:

-------------------------------------------------- -----------------------------------------
INFO Reading configuration /config/home_assistant_glow.yaml ...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup / down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup / down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C ++ source ...
INFO Compiling app ...
Processing home-assistant-glow (board: nodemcu-32s; framework: arduino; platform: platformio / espressif32 @ 3.3.2)
-------------------------------------------------- ------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
| - <AsyncTCP-esphome> 1.2.2
| - <WiFi> 1.0
| - <FS> 1.0
| - <Update> 1.0
| - <ESPAsyncWebServer-esphome> 2.1.0
| | - <AsyncTCP-esphome> 1.2.2
| - <DNSServer> 1.1.0
| - <ArduinoJson-esphomelib> 5.13.3
| - <Improv> 1.0.0
| - <ESPmDNS> 1.0
Compiling .pioenvs / home-assistant-glow / src / esphome / components / api / api_connection.cpp.o
Compiling .pioenvs / home-assistant-glow / src / esphome / components / api / api_frame_helper.cpp.o
Compiling .pioenvs / home-assistant-glow / src / esphome / components / api / api_pb2.cpp.o
Compiling .pioenvs / home-assistant-glow / src / esphome / components / api / api_pb2_service.cpp.o

-------------------------------------------------- ---------------------------------
And from there the process is totally blocked, I have tried to leave it for hours, in case it was a very long process, but I have not achieved absolutely anything after 7-8 hours of waiting ...

klaasnicolaas, 2021-12-20:
Your logs look normal and I don't think there is anything wrong with the yaml file. Compiling is usually done in 5 minutes, which should not take hours. Alternatively you could use ESP web tools via this website, then you install the firmware via the browser on your ESP instead of compiling everything yourself. https://klaasnicolaas.github.io/home-assistant-glow

Helmasynti, 2021-12-21:
Have you tried removing the LED from GPIO2 and GPIO4? My ESP could not enter flash mode before I did that.

klaasnicolaas, 2021-12-21:
That would also be a good solution, although the problem is already with compiling the firmware. Which version of ESPHome are you using?

sfritz67, 2021-12-24:
I had same issue on RPi3, crash when compiling, no response... it came from not enough capacity.
I had a RPI4 2Gb for another project, tried with it, and it worked.

caaveiro, 2021-12-25:
Hello.
It seems that the problem is memory capacity (I have an orange pi with 512 Mb of ram). Searching in forums I have found people had the same problem and it always seemed to be the same problem: lack of ram memory.
I have solved it temporarily by creating a debian virtual machine in windows 10 with the Hyper-v software, it allowed me to compile perfectly.
On the other hand, I have ordered the purchase of a Rpi4 2gb ram to transfer the installation of home assistan and esphome.

caaveiro, 2022-01-14:
I write this message to inform that after installing the Rpi 4 and testing the system everything works OK ...
I had an Orange PI Lit 512 Mb of ram and it seems that it is very little ram memory for projects that have a considerable size like this ... for simple projects it worked correctly ...

closed 2022-02-20 by github-actions[bot]

smulle48, 2022-04-01:
My rpi4 with 2gb ram kept locking up during compile.

I've found out that the Debian supervised install om HA did not have any swap file. After following this guide, it compiled just fine and didn't crash :-)

https://linuxize.com/post/how-to-add-swap-space-on-debian-10/