klaasnicolaas/home-assistant-glow #102: Not registering anything?

thehiddentruth, 2021-12-24:
I'm trying to understand, why it's not registering any data.

When I use my phones flash to trigger the photodiode, the D0 led blinks, but nothing gets registeret, neither in the logs nor in HA.

**Setup looks like this**
- Wemos D1 Mini Lite
- 3.3V -> VCC
- GND -> GND
- D2 -> D0

**My ESPHome yaml looks like this**

`substitutions:
  device_name: energy
  friendly_name: House
  device_description: "Measure your energy consumption with the pulse LED on your smart meter"
  pulse_pin: GPIO4
  status_led: GPIO5

  # imp/kWh rate ⬇ #
  pulse_rate: '1000'

  # Webserver credentials ⬇ #
  web_username: 'glow'
  web_password: 'esphome'

esphome:
  name: '${device_name}'

esp8266:
  board: d1_mini_lite

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "2419185c2ad530d793020704c6fb5aab"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Energy Fallback Hotspot"
    password: "tHerI73VhJbd"

captive_portal:

output:
  # - platform: gpio
  #   pin: GPIO5
  #   id: output_blue
  - platform: gpio
    pin: GPIO2
    id: output_red
  - platform: gpio
    pin: GPIO4
    id: output_green

light:
  - platform: binary
    internal: true
    id: led_red
    name: Red
    output: output_red

# Status LED for connection
status_led:
  pin:
    # Blue LED
    number: ${status_led}

# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: '${friendly_name} - ESPHome Version'
  - platform: wifi_info
    ip_address:
      name: '${friendly_name} - IP Address'
      icon: mdi:wifi
    ssid:
      name: '${friendly_name} - Connected SSID'
      icon: mdi:wifi-strength-2

sensor:
  - platform: pulse_meter
    id: energy_pulse_meter
    name: '${friendly_name} - Power consumption'
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    icon: mdi:flash-outline
    accuracy_decimals: 0
    pin: ${pulse_pin}
    # internal_filter: 100ms
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
    filters:
      # multiply value = (60 / imp value) * 1000
      # - multiply: 60
      - lambda: return x * ((60.0 / ${pulse_rate}) * 1000.0);
    total:
      name: '${friendly_name} - Total energy'
      unit_of_measurement: 'kWh'
      icon: mdi:circle-slice-3
      state_class: total_increasing
      device_class: energy
      accuracy_decimals: 3
      filters:
        # multiply value = 1 / imp value
        # - multiply: 0.001
        - lambda: return x * (1.0 / ${pulse_rate});
`

Any idea why nothing gets registered?

klaasnicolaas, 2021-12-25:
You now have the pulse led and status led both on GPIO4, set the pulse led to D6 (GPIO12) 😉

thehiddentruth, 2021-12-25:
> You now have the pulse led and status led both on GPIO4, set the pulse led to D6 (GPIO12) 😉 

You're right. But I originally had it on D6 (GPIO12) as the documentation said, but still the exact same thing? No readings at all. I changed there GPIO because I thought that the D6 was damaged or something..

I also checked to make sure that the photodiode is soldered correctly, as per documentation.

klaasnicolaas, 2021-12-26:
Have you tried another cable? Or another ESP?

thehiddentruth, 2021-12-27:
> Have you tried another cable? Or another ESP?

Yes, I have tried both with an wemos d1 mini and an esp32, and with 3 different micro USB cables. Exact same.. no readings at all

klaasnicolaas, 2021-12-27:
What color is the pulse LED? Your issue might be similar to this one #62?

thehiddentruth, 2021-12-28:
> What color is the pulse LED? Your issue might be similar to this one #62?

I haven't connected the RGB diode, as I don't have one.

Could that be the issue? That I haven't connected an RGB diode?

klaasnicolaas, 2021-12-28:
I don't think that has anything to do with each other.

closed 2022-02-04 by github-actions[bot]