klaasnicolaas/home-assistant-glow #109: Can a Photo Resistor be used instead of a Photodiode?

dweng0, 2022-01-05:
Hi, Resistors tend to be more sensitive to light, that said, would a breakout board with a resistor instead of a diode be a viable option? can the values in the script be tweaked to accommodate for it?

Jay

klaasnicolaas, 2022-01-05:
You can use a photoresistor with the glow, it is recommended to use the [internal_filter](https://github.com/klaasnicolaas/home-assistant-glow/blob/main/home_assistant_glow.yaml#L133) option because of higher sensitivity.

dweng0, 2022-01-06:
Just looking at the theory here, but [a simple photoresistor](https://nl.banggood.com/search/photoresistor.html?from=nav) hooked from an esp32board (5v) to a [5kohm resistor](https://www.ebay.co.uk/itm/263894047668) and back to the board (gnd).

We can then measure the connection between the photoresistor and the fixed resistor by connecting a wire to their connection and giving it to an Analogue input, for example (P12).

couple this with a [small breadboard](https://www.ebay.co.uk/itm/284000081310)
and some [jumper lead wires](https://www.ebay.co.uk/itm/261998236967?epid=1342511679&hash=item3d004fd527:g:98MAAOSwnK9ZXMIP)

And you have an entry-level friendly reader and it doesn't 'matter which way round the resistors are 

(Had a photodiode that was soldered the wrong way, and I suck at soldering!).

raforo, 2022-01-06:
I'm interested in alternatives to the recommended photodiodes, I've bought three at Banggood, they all arrived with the polarity changed, I soldered correctly, but I cannot adjust the sensitivity to the meter's LED, it seems that it is very dim and goes from giving a continuous signal in DO to give no signal in DO when moving the variable resistor.
Everything else works perfectly, but I cannot adjust the sensitivity of the Led , a strong light like a flashlight It's detected, but the LED does not.
Any advice? Thanks

dweng0, 2022-01-06:
Is the solder connection good? a weak signal can mean that the circuit is not sound, likely cause being the soldering or the pins being loose

raforo, 2022-01-06:
> Just looking at the theory here, but [a simple photoresistor](https://nl.banggood.com/search/photoresistor.html?from=nav) hooked from an esp32board (5v) to a [5kohm resistor](https://www.ebay.co.uk/itm/263894047668) and back to the board (gnd).
> 
> We can then measure the connection between the photoresistor and the fixed resistor by connecting a wire to their connection and giving it to an Analogue input, for example (P12).
> 
> couple this with a [small breadboard](https://www.ebay.co.uk/itm/284000081310) and some [jumper lead wires](https://www.ebay.co.uk/itm/261998236967?epid=1342511679&hash=item3d004fd527:g:98MAAOSwnK9ZXMIP)
> 
> And you have an entry-level friendly reader and it doesn't 'matter which way round the resistors are
> 
> [esp32] [5v] -> [photoresistor] - - . - - > [5kohm resistor] -> [esp32][gnd] <--- This is just a simple circuit
> 
> [esp32] [p12] <-------------------- ^ <--- this connection is what we use to measure the light resistance
> 
> We can then use the code provided in this repo, but tweaked to the correct pin
> 
> (Had a photodiode that was soldered the wrong way, and I suck at soldering!).

Have you made and tested your scheme? 
I don't understand where to connect pin 12 of esp32? 
Would it work connecting to the 3.3V output instead of the 5V output of the esp32?

dweng0, 2022-01-06:
> I don't understand where to connect pin 12 of esp32?
GPIO12 
[image]

You can technically connect it to any ADC pin (Analogue to digital conversion) I picked 12 because that's what the code in this repo reads its input from. Otherwise, you'll have to modify the code to read from a different pin. 

>Would it work connecting to the 3.3V output instead of the 5V output of the esp32?
3.3v would require a 3.3vresistor instead of the 5v one, but in theory, it will work, the values would of course be different, and 
@klassnicolas mentioned that the following line would need to be uncommented in the scheme to accommodate the additional sensitivity:

https://github.com/klaasnicolaas/home-assistant-glow/blob/main/home_assistant_glow.yaml#L133

raforo, 2022-01-06:
> 

> > I don't understand where to connect pin 12 of esp32?
> 
> GPIO12, sometimes called P12 or just 12 (pin 12) on esp32 boards 

Thank you very much, I know where pin 12 is, I mean where to connect the other side, reviewing your scheme, I think is in the joining between the photodiode and the 5k ohm resistor. I apologize for my english

raforo, 2022-01-12:
I've replaced the photodiode by an LDR on the Bangood board and it has started working properly, now it detects the pulse of the red LED of the meter without problem.

Thanks

closed 2022-02-19 by github-actions[bot]