klaasnicolaas/home-assistant-glow #113: Sensor for current

robinelfving, 2022-01-09:
Would it be possible to implement sensor to show the current Amps aswell as the watts?

klaasnicolaas, 2022-01-09:
From what exactly? the ESP?

robinelfving, 2022-01-09:
Sorry, i know it cannot be a messarued value. But I want it as än calculated value based on the Watt reading. My coding isnt good enough. 

I = P * V
I = Amp 
P = Watt
V = Volt

Would that be possible?

klaasnicolaas, 2022-01-09:
I think this is quite out of scope. It would be better to tackle this with CT clamps and then have the formulas calculated with template sensors in Home Assistant.

robinelfving, 2022-01-10:
Ok, thank you for the reply!

closed 2022-01-10 by robinelfving

robinelfving, 2022-01-25:
Just for information.

I did solve this like this:
>   - platform: template
    name: "${friendly_name} - Amps"
    id: sensor_total_amps
    unit_of_measurement: 'A'
    icon: mdi:current-ac
    accuracy_decimals: 1
    lambda: |-
      return id(sensor_energy_pulse_meter).state / 230;
    update_interval: 1s