klaasnicolaas/home-assistant-glow #114: Not compiling

nzneil22, 2022-01-10:
Hello, i am trying to install the home assistant glow binary using esphome on my HA instance. 

I keep getting thee following error:

[image]

Any insights? I am new to this 

Neil

klaasnicolaas, 2022-01-10:
Which ESPHome version are you using? I don't experience any problems compiling the binary myself 🤔

nzneil22, 2022-01-10:
I am using version 2021.12.3

nzneil22, 2022-01-14:
Thanks for your suggestion, yet I have already tried. It has not worked either. 

I am running Home Assistant OS on an RPI 4.

nzneil22, 2022-01-14:
I'm sorry what YAML config? 

I created an esphome file where I have the following code:
https://controlc.com/5b4452ca

klaasnicolaas, 2022-01-14:
I found some threads, what could be interesting:
- https://github.com/esphome/issues/issues/1386
- https://community.home-assistant.io/t/lower-the-number-of-cc1plus-compilation-threads/238911/8

On what kind of device are you running Home Assistant?

nzneil22, 2022-01-16:
> I found some threads, what could be interesting:
>
>    - esphome/issues#1386 <https://github.com/esphome/issues/issues/1386>
>    -
>    https://community.home-assistant.io/t/lower-the-number-of-cc1plus-compilation-threads/238911/8
>
> On what kind of device are you running Home Assistant?

I will look into them thanks.

I am running a Home Assistant OS instance on a Raspberry Pi 4.

- Neil

nzneil22, 2022-01-17:
Hello @klaasnicolaas 

For some reason, I compiled the YAML file on a completely fresh install of home assistant os and it worked without any flaws.

I guess my esphome install was corrupt. 

Thank you for your help.

Neil

closed 2022-01-17 by nzneil22