klaasnicolaas/home-assistant-glow #128: Maximum cable length from photodiode to ESP?

EspressoNerd, 2022-02-03:
Due to low WiFi signal strength I need to place the ESP away from the photodiode. What is the maximum cable length for the connection of the ESP and the photodiode? I am aware that this isn't a fixed length (0,9m is ok, 1 m isn't), but at least an order of magnitude would be good; e.g. is 2 m ok?

Kimble8650, 2022-02-07:
I'd say, that 2m is probably fine. Especially if you don't have any large sources of EM-interference nearby. If voltage drop becomes an issue (it prob. won't), just use a thicker wire.

EspressoNerd, 2022-02-09:
Thank you, great news.

Kimble8650, 2022-02-23:
To elaborate a bit further on the topic; some users have reported spikes in the reported power consumption, which might be due to EM-interference caused by the power meter itself. Many power meters nowadays have remote consumption reporting, and send their data over an RF link. 

So stuff you can do to mitigate this is to route your cable avoiding the antennae in the power meter, use a shielded cable, place a filtering capacitor between the impulse-pin and ground and add some kind of internal filter value to the configuration (maybe 100ms or so. depending on the impulse rate of your power meter).

closed 2022-04-02 by github-actions[bot]