klaasnicolaas/home-assistant-glow #132: S0 to Glow. problem solved

zorrosweden, 2022-02-10:
I have solved the problem with the S0 output to Glow.
Solder off the photodiode and connect 2 cables as the picture shows works perfectly

[image]

klaasnicolaas, 2022-02-10:
Didn't think it would be so easy, are the readings good? 😄 If so, it might be interesting to write something about this in the documentation, such as wiring and example code (if it differs) because that means that the Glow can be used more diversely 🥳

zorrosweden, 2022-02-14:
https://www.dropbox.com/s/zllvf80ehd5mynn/20220210_080032%20%281%29.mp4?dl=0

msteve72, 2022-02-15:
Hi
Please, I also have a problem because in one day I have a difference of about 20 kW because I have it set
I set the HA parameters as I have the image in the clock
544.42, but the next day I have 573,285 in Homeassistent
Your settings work well and these values are the same between the clock and the Homeassistant 
[image]
[image]