klaasnicolaas/home-assistant-glow #139: Add Uptime Sensor

pschneider87, 2022-03-03:
I find it useful to have an Uptime Sensor in HA Glow. I've copied right away from the ESPHome-Page from the "Human readable sensor" Section [¶](https://esphome.io/components/sensor/uptime.html#human-readable-sensor)

I'm not so familiar with GitHub, so I'm afraid I can't make a PR :-/
But maybe somebody can quickly copy/paste it into the project, is quite nice to know about it:

[image]

pschneider87, 2022-03-07:
added https://github.com/klaasnicolaas/home-assistant-glow/pull/142
don't know if @klaasnicolaas needs to do something to make the tests verify the code :-/

klaasnicolaas, 2022-03-07:
What about this uptime sensor: https://github.com/klaasnicolaas/smarty-homeassistant-config/blob/master/esphome/sensors/uptime_timestamp.yaml

pschneider87, 2022-03-07:
I'll give that a try. The idea sounds good to only send it once.

klaasnicolaas, 2022-03-10:
And?

klaasnicolaas, 2022-04-18:
Still something that is on my list to add 🤔

mcchots, 2023-02-07:
Doesn't this [uptime sensor](https://esphome.io/components/sensor/uptime.html) solve this issue?

jessereynolds, 2023-05-22:
Yep, I've been using the uptime sensor @mcchots mentions with success. Added the following to the esphome build definition yaml file: 

```
  - platform: uptime
    name: Uptime
    id: sensor_uptime
    update_interval: 60s
```

spanznet, 2024-01-30:
Uptime is great, but rather than have to calculate backwards in time to work out when it last booted, I prefer to add one that updates once on each boot when the time syncs:

```
text_sensor:
  - platform: template
    name: "Last Boot"
    id: device_last_boot
    icon: mdi:clock
    entity_category: diagnostic

time:
  - platform: homeassistant
    timezone: "Australia/Hobart"
    id: homeassistant_time
    on_time_sync:
      then:
        - if:
            condition:
              lambda: 'return id(device_last_boot).state == "";'
            then:
              - text_sensor.template.publish:
                  id: device_last_boot
                  state: !lambda 'return id(homeassistant_time).now().strftime("%a %d %b %Y - %I:%M:%S %p");'
```

Which gives you:

[image]