klaasnicolaas/home-assistant-glow #140: Daily Energy doesn't reset

srslykev, 2022-03-05:
Hi there, 
so far, everything is working great except the daily energy meter. It won't reset at midnight. I'm using the regular code with no changes apart from the led on pulse part:
 
```
    on_value:
      then:
        - light.turn_on:
            id: led_red
        - delay: 0.5s
        - light.turn_off:
            id: led_red
```

Any idea what could be wrong?

ghitchen, 2022-03-05:
I had the same problem. If you are using sntp for the timeplatform then you need DNS entries in your wifi: section. I changed to using:

time:
  - platform: homeassistant
    id: my_time

as this doesn't need DNS or Internet access from the ESP.

srslykev, 2022-03-05:
Great, thanks! Will try that 😊

klaasnicolaas, 2022-03-08:
And has it helped? 😄

srslykev, 2022-03-08:
Sorry, Yes, works great, now! 😊

klaasnicolaas, 2022-03-08:
In that case I will close the issue, Maybe adding this also to the FAQ 🤔

ghitchen, 2022-03-09:
Would it not be better to change the time: platform: to homeassistant  in the yaml ?

klaasnicolaas, 2022-03-09:
The problem I run into is that when I use that platform my daily entity will not reset, which is why it has not yet been modified in the yaml file.

klaasnicolaas, 2022-03-10:
Strange, the reset works fine now. So let's just modify it and name the sntp as an alternative.

ghitchen, 2022-03-10:
> Strange, the reset works fine now. So let's just modify it and name the sntp as an alternative.

Probably worth adding that if using sntp as an alternative then if you are using Manual IPs make sure to configure a DNS Server (dns1, dns2) or use only IP addresses for the NTP servers.

closed 2022-03-14 by klaasnicolaas