klaasnicolaas/home-assistant-glow #147: Total Energy keeps resetting

niklerus, 2022-03-17:
Really great project!
Everything works fine except that the Energy total keeps resetting to 0.
It can work for 30 minutes to a few hours but always ends up resetting.
I moved some of the pins since I got some warning about GPIO 2, 4 and 12 were strapping pins and edit the yaml accordingly, didn't help (did not get the strapping warning though).
Also commented out the Reset Total button function, thinking there might be some random triggering of that, didn't help.
Could it be related to the actual board?
I don't really get what the lines in all yaml files does?;
_esp32:
   board: esp32dev_

I got esp32dev from some other example code, the actual chip says **espressif wroom 32D** but I couldn't find that on the EspHome homepage list over different boards.
But does it really matter? As long as it is an Esp32 the pins from the chip are gonna be the same, right?
The layout of the pins on the board might differ but as long as you connect to the correct pins it should work?
Or is this the reason for my issue, I haven't got the correct board?
And if so, what should I enter as board for the Wroom 32D?

Optsoul, 2022-03-19:
Got the same issue just now. The exact error is the '[E][api:127]: No client connected to API. Rebooting...'.

[D][sensor:125]: 'House - Total Energy': Sending state 0.43400 kWh with 3 decimals of accuracy
[D][sensor:125]: 'House - Total Energy': Sending state 0.43400 kWh with 3 decimals of accuracy
[D][light:035]: 'Red' Setting:
[D][light:046]:   State: OFF
[E][api:127]: No client connected to API. Rebooting...
[I][app:127]: Forcing a reboot...
[D][esp32.preferences:114]: Saving preferences to flash...
[D][light:035]: 'Red' Setting:
[D][light:046]:   State: OFF
[D][sensor:125]: 'House - Power Consumption': Sending state 602.20807 W with 0 decimals of accuracy
[D][sensor:125]: 'House - Power Consumption': Sending state 602.20807 W with 0 decimals of accuracy
[D][light:035]: 'Red' Setting:

klaasnicolaas, 2022-03-19:
It will reboot if home assistant is not connected to the node. So maybe you should re-add to home assistant.

niklerus, 2022-03-19:
But why is the Daily Energy persistant through a reboot?
I actually think mine is OK now, installed it as a nodemcu-32s board and it seems to have made the problem to go away, been up since beginning of the 18th.
Also, I don't get the strapping warnings anymore, after I reassigned the pins away from GPIO2 and GPIO4.
Shouldn't they be avoided?

Eddie46848, 2022-03-25:
Hi

I had this problem, see link https://github.com/klaasnicolaas/home-assistant-glow/issues/110

klaasnicolaas, 2022-04-01:
Using other pins solves the warnings, but would not make much difference to the functioning of the glow. Maybe I should see if I can't better use other GPIO pins 🤔

kazlon1, 2022-04-04:
I found solution in my case, maybe it will help other. I had the problem with resterting daily and total count, but in my case it was always after esp wanted save preferences to flash:

[D][esp32.preferences:114]: Saving preferences to flash... 
INFO 192.168.2.254: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer 
INFO Disconnected from ESPHome API for 192.168.2.254 
WARNING Disconnected from API 
INFO 192.168.2.254: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer INFO Successfully connected to 192.168.2.254

So after long research i found this:

preferences:
  flash_write_interval: 24hours

After adding the above mentioned code, my problems are gone.

klaasnicolaas, 2022-04-04:
Isn't that more of a consequence because it has lost the connection with the ESPHome API? At least that's what it seems in the log you share.

kazlon1, 2022-04-04:
> Isn't that more of a consequence because it has lost the connection with the ESPHome API? At least that's what it seems in the log you share. In any case, shouldn't be necessary to set that up. I've been running the stuff for over a year now and never had any problems.

You are probably right. Previously, I had random disconnects even every 40 minutes. Now it has been working without a problem for 4 days. Unfortunately I can't use Wi-Fi repeater or longer USB cable to bypass connection problems.

niklerus, 2022-04-04:
Still don't understand why daily consumption survives a reboot but not the total consumption?
If I had a choice I would prefer the total to be the persistent value.
Is there a way to set that up?

klaasnicolaas, 2022-04-04:
Because they are 2 different components where one has the functionality and the other does not. It is best to submit a feature request to ESPHome if there is not already one.

closed 2022-05-13 by github-actions[bot]