klaasnicolaas/home-assistant-glow #148: Multiple readings for 'Power consumption' ?

janneojala, 2022-03-21:
I just put in one piece this great gadget, however I'm getting multiple readings for 'Power consumption'. My energy meter has two leds, one with 10000 impl/kWh and a S0 led with 500 impl / kWh. I'm using the latter. RGB seems to blink accordingly to S0, but in the log I get two readings. ~ 3840.00000 W could be quite accurate, but where does the 75000.0000 W come from? 

11:01:00 [D] [light:035] Red Setting:
11:01:01 [D] [sensor:125] House - Power Consumption: Sending state 3840.00000 W with 0 decimals of accuracy
11:01:01 [D] [light:035] Red Setting:
11:01:01 [D] [light:046]   State: ON
11:01:01 [D] [sensor:125] House - Daily Energy: Sending state 56.57299 kWh with 3 decimals of accuracy
11:01:01 [D] [sensor:125] House - Total Energy: Sending state 50.29600 kWh with 3 decimals of accuracy
11:01:01 [D] [sensor:125] House - Power Consumption: Sending state 75000.00000 W with 0 decimals of accuracy
11:01:01 [D] [light:035] Red Setting:
11:01:01 [D] [sensor:125] House - Daily Energy: Sending state 56.57492 kWh with 3 decimals of accuracy
11:01:01 [D] [sensor:125] House - Total Energy: Sending state 50.29800 kWh with 3 decimals of accuracy
11:01:02 [D] [light:035] Red Setting:
11:01:02 [D] [light:046]   State: OFF
11:01:02 [D] [light:035] Red Setting:
11:01:03 [D] [sensor:125] House - Power Consumption: Sending state 3852.32764 W with 0 decimals of accuracy
11:01:03 [D] [light:035] Red Setting:
11:01:03 [D] [light:046]   State: ON
11:01:03 [D] [sensor:125] House - Daily Energy: Sending state 56.57694 kWh with 3 decimals of accuracy
11:01:03 [D] [sensor:125] House - Total Energy: Sending state 50.30000 kWh with 3 decimals of accuracy
11:01:03 [D] [sensor:125] House - Power Consumption: Sending state 75000.00000 W with 0 decimals of accuracy
11:01:03 [D] [light:035] Red Setting:
11:01:03 [D] [sensor:125] House - Daily Energy: Sending state 56.57890 kWh with 3 decimals of accuracy
11:01:03 [D] [sensor:125] House - Total Energy: Sending state 50.30200 kWh with 3 decimals of accuracy
11:01:03 [D] [sensor:125] House - Total Energy: Sending state 50.30200 kWh with 3 decimals of accuracy
11:01:04 [D] [light:035] Red Setting:
11:01:04 [D] [light:046]   State: OFF
11:01:04 [D] [light:035] Red Setting:
11:01:05 [D] [sensor:125]

klaasnicolaas, 2022-03-21:
Are those diodes next to each other? Is there a chance that the glow is measuring both LEDs at the same time?

janneojala, 2022-03-23:
The diodes are quite close each other. I blocked the 10000 imp diode with tape and added throttle_average: 10s to filters. Now I get only one reading and it seams accurate.

klaasnicolaas, 2022-03-27:
Can we close this issue then?

closed 2022-03-31 by janneojala