klaasnicolaas/home-assistant-glow #155: Energy reset on reboot

smulle48, 2022-03-31:
Hi

Is it correct that this can't store the current value on the module?

Just had to swap the case I've printed, and when I powered it on again, both House - Daily Energy and House - Total Energy was back to zero.

Just to be sure, I let it run a few minutes and rebooted it again, and again it went back to zero.

It this correct? or can it somehow be fixed to store the value?

I'm running an WeMos D1 mini clone if that makes any difference

klaasnicolaas, 2022-04-01:
The day entity should (by default) always remember its last value and show it again after a reboot or firmware flash. Unfortunately the total entity does not have that option and will reset every time on reboot or firmware flash.

klaasnicolaas, 2022-04-01:
Does that also happen with a firmware flash? Unfortunately I can't test with my own (production) Glow because they have replaced the door cylinders, so I can't reach it permanently 😅

closed 2022-05-10 by github-actions[bot]