klaasnicolaas/home-assistant-glow #156: Minimal case

smulle48, 2022-04-09:
First of all, an amazing project :-)

I thought I would share my minimal case for this project.

As the diode on my meter was just next to the display, I could not use the original case, as it blocked part of the display.

The case for the d1 mini have room for pinheader, and as the sensor case, secures the wires/usb cable so you don't accidentially pull them out.

Both cases are secured with double sided carpet tape and hold perfectly.

https://www.thingiverse.com/thing:5347120

klaasnicolaas, 2022-04-27:
Do you have some pictures of how it was finally installed on your meter? (possibly adding images to thingiverse)

smulle48, 2022-04-27:
> Do you have some pictures of how it was finally installed on your meter? (possibly adding images to thingiverse)

Sure. Here are some :-)
[image]
[image]
[image]
[image]

closed 2022-06-05 by github-actions[bot]